open-mmlab/OpenPCDet #200: error in Quick Demo

Jsijon, 2020-08-01:
i am running demo.py and the output is this:

2020-08-01 17:44:26,046   INFO  ==> Done (loaded 16/367)
2020-08-01 17:44:26,071   INFO  Visualized sample index: 	1
Segmentation fault (core dumped)

what is wrong ?

thank you

sshaoshuai, 2020-08-01:
Could you train or test the model first and see if it works? 
The segmentation fault means that there may exist some errors in your compilation.

Jsijon, 2020-08-01:
i tried to train and i can't. see issue https://github.com/open-mmlab/OpenPCDet/issues/201

Jsijon, 2020-08-02:
issue  https://github.com/open-mmlab/OpenPCDet/issues/201 is resolved, ican train now.
but still i can't run quick demo.
how i can fix the errors in compilation?

sshaoshuai, 2020-08-03:
Could you also post your detailed error log to here?

Jsijon, 2020-08-03:
2020-08-03 20:04:21,784   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2020-08-03 20:04:21,941   INFO  Total number of samples: 	1
2020-08-03 20:04:24,733   INFO  ==> Loading parameters from checkpoint pv_rcnn_8369.pth to CPU
2020-08-03 20:04:25,461   INFO  ==> Done (loaded 367/367)
2020-08-03 20:04:25,483   INFO  Visualized sample index: 	1
Segmentation fault (core dumped)

Jsijon, 2020-08-03:
also i train with poinpillar sucessfully but with SECOND same error.

(base) john@john-desktop:~/OpenPCDet/tools$ python train.py --cfg_file cfgs/kitti_models/second.yaml --epochs 1 --batch_size 2
2020-08-03 20:28:30,501   INFO  **********************Start logging**********************
2020-08-03 20:28:30,501   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-08-03 20:28:30,501   INFO  cfg_file         cfgs/kitti_models/second.yaml
2020-08-03 20:28:30,501   INFO  batch_size       2
2020-08-03 20:28:30,501   INFO  epochs           1
2020-08-03 20:28:30,501   INFO  workers          8
2020-08-03 20:28:30,501   INFO  extra_tag        default
2020-08-03 20:28:30,501   INFO  ckpt             None
2020-08-03 20:28:30,501   INFO  pretrained_model None
2020-08-03 20:28:30,501   INFO  launcher         none
2020-08-03 20:28:30,501   INFO  tcp_port         18888
2020-08-03 20:28:30,502   INFO  sync_bn          False
2020-08-03 20:28:30,502   INFO  fix_random_seed  False
2020-08-03 20:28:30,502   INFO  ckpt_save_interval 1
2020-08-03 20:28:30,502   INFO  local_rank       0
2020-08-03 20:28:30,502   INFO  max_ckpt_save_num 30
2020-08-03 20:28:30,502   INFO  merge_all_iters_to_one_epoch False
2020-08-03 20:28:30,502   INFO  set_cfgs         None
2020-08-03 20:28:30,502   INFO  max_waiting_mins 0
2020-08-03 20:28:30,502   INFO  start_epoch      0
2020-08-03 20:28:30,502   INFO  save_to_file     False
2020-08-03 20:28:30,502   INFO  cfg.ROOT_DIR: /home/john/OpenPCDet
2020-08-03 20:28:30,502   INFO  cfg.LOCAL_RANK: 0
2020-08-03 20:28:30,502   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-08-03 20:28:30,502   INFO  
cfg.DATA_CONFIG = edict()
2020-08-03 20:28:30,502   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-08-03 20:28:30,502   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2020-08-03 20:28:30,502   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2020-08-03 20:28:30,502   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-08-03 20:28:30,502   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-08-03 20:28:30,503   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-08-03 20:28:30,503   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:20', 'Pedestrian:15', 'Cyclist:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-08-03 20:28:30,503   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2020-08-03 20:28:30,503   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2020-08-03 20:28:30,503   INFO  
cfg.MODEL = edict()
2020-08-03 20:28:30,503   INFO  cfg.MODEL.NAME: SECONDNet
2020-08-03 20:28:30,503   INFO  
cfg.MODEL.VFE = edict()
2020-08-03 20:28:30,503   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2020-08-03 20:28:30,504   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2020-08-03 20:28:30,504   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2020-08-03 20:28:30,504   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2020-08-03 20:28:30,504   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2020-08-03 20:28:30,504   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2020-08-03 20:28:30,504   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2020-08-03 20:28:30,504   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2020-08-03 20:28:30,504   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2020-08-03 20:28:30,504   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
[... 188 lines cut ...]
      (0): Sequential(
        (0): ConvTranspose2d(128, 256, kernel_size=(1, 1), stride=(1, 1), bias=False)
        (1): BatchNorm2d(256, eps=0.001, momentum=0.01, affine=True, track_running_stats=True)
        (2): ReLU()
      )
      (1): Sequential(
        (0): ConvTranspose2d(256, 256, kernel_size=(2, 2), stride=(2, 2), bias=False)
[... cut ...]
    (conv_cls): Conv2d(512, 18, kernel_size=(1, 1), stride=(1, 1))
    (conv_box): Conv2d(512, 42, kernel_size=(1, 1), stride=(1, 1))
    (conv_dir_cls): Conv2d(512, 12, kernel_size=(1, 1), stride=(1, 1))
  )
  (point_head): None
  (roi_head): None
)
2020-08-03 20:28:32,534   INFO  **********************Start training kitti_models/second(default)**********************
epochs:   0%|                                             | 0/1 [00:00<?, ?it/sSegmentation fault (core dumped)                        | 0/1856 [00:00<?, ?it/s]

Jsijon, 2020-08-03:
and bellow the log of the compilation:

running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.pointnet2.pointnet2_stack.pointnet2_stack_cuda' extension
Emitting ninja build file /home/john/OpenPCDet/build/temp.linux-x86_64-3.8/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/9] c++ -MMD -MF /home/john/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.o.d -pthread -B /home/john/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/john/anaconda3/lib/python3.8/site-packages/torch/include -I/home/john/anaconda3/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/john/anaconda3/lib/python3.8/site-packages/torch/include/TH -I/home/john/anaconda3/lib/python3.8/site-packages/torch/include/THC -I/home/john/anaconda3/include/python3.8 -c -c /home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp -o /home/john/OpenPCDet/build/temp.linux-x86_64-3.8/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
/home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp: In function ‘int furthest_point_sampling_wrapper(int, int, int, at::Tensor, at::Tensor, at::Tensor)’:
/home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp:10:15: warning: ‘at::DeprecatedTypeProperties& at::Tensor::type() const’ is deprecated: Tensor.type() is deprecated. Instead use Tensor.options(), which in many cases (e.g. in a constructor) is a drop-in replacement. If you were using data from type(), that is now available from Tensor itself, so instead of tensor.type().scalar_type(), use tensor.scalar_type() instead and instead of tensor.type().backend() use tensor.device(). [-Wdeprecated-declarations]
   10 |   if (!x.type().is_cuda()) { \
      |               ^
/home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp:21:24: note: in expansion of macro ‘CHECK_CUDA’
   21 | #define CHECK_INPUT(x) CHECK_CUDA(x);CHECK_CONTIGUOUS(x)
      |                        ^~~~~~~~~~
/home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp:27:5: note: in expansion of macro ‘CHECK_INPUT’
   27 |     CHECK_INPUT(points_tensor);
      |     ^~~~~~~~~~~
In file included from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/ATen/Tensor.h:11,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/ATen/Context.h:4,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/ATen/ATen.h:5,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from /home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp:1:
/home/john/anaconda3/lib/python3.8/site-packages/torch/include/ATen/core/TensorBody.h:262:30: note: declared here
  262 |   DeprecatedTypeProperties & type() const {
      |                              ^~~~
/home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp:10:15: warning: ‘at::DeprecatedTypeProperties& at::Tensor::type() const’ is deprecated: Tensor.type() is deprecated. Instead use Tensor.options(), which in many cases (e.g. in a constructor) is a drop-in replacement. If you were using data from type(), that is now available from Tensor itself, so instead of tensor.type().scalar_type(), use tensor.scalar_type() instead and instead of tensor.type().backend() use tensor.device(). [-Wdeprecated-declarations]
   10 |   if (!x.type().is_cuda()) { \
      |               ^
/home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp:21:24: note: in expansion of macro ‘CHECK_CUDA’
   21 | #define CHECK_INPUT(x) CHECK_CUDA(x);CHECK_CONTIGUOUS(x)
      |                        ^~~~~~~~~~
/home/john/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp:28:5: note: in expansion of macro ‘CHECK_INPUT’
   28 |     CHECK_INPUT(temp_tensor);
      |     ^~~~~~~~~~~
In file included from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/ATen/Tensor.h:11,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/ATen/Context.h:4,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/ATen/ATen.h:5,
                 from /home/john/anaconda3/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
[... cut ...]
Searching for llvmlite==0.33.0+1.g022ab0f
Best match: llvmlite 0.33.0+1.g022ab0f
Adding llvmlite 0.33.0+1.g022ab0f to easy-install.pth file

Using /home/john/anaconda3/lib/python3.8/site-packages
Searching for future==0.18.2
Best match: future 0.18.2
Adding future 0.18.2 to easy-install.pth file
Installing futurize script to /home/john/anaconda3/bin
Installing pasteurize script to /home/john/anaconda3/bin

Using /home/john/anaconda3/lib/python3.8/site-packages
Finished processing dependencies for pcdet==0.3.0+96a6092

sshaoshuai, 2020-08-04:
What's your version of GCC, CUDA, PyTorch and spconv?

Jsijon, 2020-08-04:
gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
Cuda compilation tools, release 10.1, V10.1.243
Pytorch 1.5.0
spconv 1.2.1

sshaoshuai, 2020-08-05:
As mentioned in the original repo of [spconv](https://github.com/traveller59/spconv), you need use their docker to run spconv 1.2 with pytorch 1.5.

Jsijon, 2020-08-09:
i still have the same problem with previous version of pytorch

i found  the line that produces the error
in the demo.py is the line:  pred_dicts, _ = model.forward(data_dict)
and  in the pv_rcnn.py in detectors folder is the line :
for cur_module in self.module_list:
            batch_dict = cur_module(batch_dict)
the second time it iterates

any suggestions ?

adibyhy, 2020-08-13:
@Jsijon I solved this issue by downgrading spconv version from 1.2 to [1.0 commit 8da6f96](https://github.com/traveller59/spconv/tree/8da6f967fb9a054d8870c3515b1b44eca2103634)

Jsijon, 2020-08-13:
@adibyhy can you tell me with which pytorch version did you use?
thank you

MartinHahner88, 2020-08-13:
maybe try this: https://github.com/traveller59/spconv/issues/152

adibyhy, 2020-08-14:
@Jsijon I use Pytorch 1.3.1 with CUDA 10.2.

closed 2020-10-21 by sshaoshuai

Petros626, 2024-10-01:
Follow this for a clean installation, demo.py shouldn't fail

https://github.com/open-mmlab/OpenPCDet/issues/688#issuecomment-2385391436