open-mmlab/OpenPCDet #201: error when training the model : AttributeError: __enter__

Jsijon, 2020-08-01:
when i try to train the model (OpenPCDet v0.3.0 )
i get this error:
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/john/OpenPCDet/tools/train_utils/train_utils.py", line 69, in train_model
    with tqdm.trange(start_epoch, total_epochs, desc='epochs', dynamic_ncols=True, leave=(rank == 0)) as tbar:
AttributeError: __enter__

what is the problem?

sshaoshuai, 2020-08-01:
Could you post the detail error log? I could help much with these few lines.

Jsijon, 2020-08-01:
2020-08-01 20:02:46,663   INFO  **********************Start logging**********************
2020-08-01 20:02:46,663   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-08-01 20:02:46,663   INFO  cfg_file         cfgs/kitti_models/pointpillar.yaml
2020-08-01 20:02:46,663   INFO  batch_size       4
2020-08-01 20:02:46,663   INFO  epochs           1
2020-08-01 20:02:46,663   INFO  workers          8
2020-08-01 20:02:46,663   INFO  extra_tag        default
2020-08-01 20:02:46,663   INFO  ckpt             None
2020-08-01 20:02:46,663   INFO  pretrained_model None
2020-08-01 20:02:46,664   INFO  launcher         none
2020-08-01 20:02:46,664   INFO  tcp_port         18888
2020-08-01 20:02:46,664   INFO  sync_bn          False
2020-08-01 20:02:46,664   INFO  fix_random_seed  False
2020-08-01 20:02:46,664   INFO  ckpt_save_interval 1
2020-08-01 20:02:46,664   INFO  local_rank       0
2020-08-01 20:02:46,664   INFO  max_ckpt_save_num 30
2020-08-01 20:02:46,664   INFO  merge_all_iters_to_one_epoch False
2020-08-01 20:02:46,664   INFO  set_cfgs         None
2020-08-01 20:02:46,664   INFO  max_waiting_mins 0
2020-08-01 20:02:46,664   INFO  start_epoch      0
2020-08-01 20:02:46,664   INFO  save_to_file     False
2020-08-01 20:02:46,664   INFO  cfg.ROOT_DIR: /home/john/OpenPCDet
2020-08-01 20:02:46,664   INFO  cfg.LOCAL_RANK: 0
2020-08-01 20:02:46,664   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-08-01 20:02:46,664   INFO  
cfg.DATA_CONFIG = edict()
2020-08-01 20:02:46,664   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-08-01 20:02:46,664   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2020-08-01 20:02:46,664   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1]
2020-08-01 20:02:46,664   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-08-01 20:02:46,664   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-08-01 20:02:46,664   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-08-01 20:02:46,664   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2020-08-01 20:02:46,664   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2020-08-01 20:02:46,664   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2020-08-01 20:02:46,664   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-08-01 20:02:46,664   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-08-01 20:02:46,665   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-08-01 20:02:46,665   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:15', 'Cyclist:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-08-01 20:02:46,665   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-08-01 20:02:46,665   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-08-01 20:02:46,665   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2020-08-01 20:02:46,665   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2020-08-01 20:02:46,665   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.16, 0.16, 4], 'MAX_POINTS_PER_VOXEL': 32, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2020-08-01 20:02:46,665   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2020-08-01 20:02:46,665   INFO  
cfg.MODEL = edict()
2020-08-01 20:02:46,665   INFO  cfg.MODEL.NAME: PointPillar
2020-08-01 20:02:46,665   INFO  
cfg.MODEL.VFE = edict()
2020-08-01 20:02:46,665   INFO  cfg.MODEL.VFE.NAME: PillarVFE
2020-08-01 20:02:46,665   INFO  cfg.MODEL.VFE.WITH_DISTANCE: False
2020-08-01 20:02:46,665   INFO  cfg.MODEL.VFE.USE_ABSLOTE_XYZ: True
2020-08-01 20:02:46,665   INFO  cfg.MODEL.VFE.USE_NORM: True
2020-08-01 20:02:46,665   INFO  cfg.MODEL.VFE.NUM_FILTERS: [64]
2020-08-01 20:02:46,665   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2020-08-01 20:02:46,665   INFO  cfg.MODEL.MAP_TO_BEV.NAME: PointPillarScatter
2020-08-01 20:02:46,665   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 64
2020-08-01 20:02:46,665   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2020-08-01 20:02:46,665   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2020-08-01 20:02:46,665   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [3, 5, 5]
2020-08-01 20:02:46,665   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [2, 2, 2]
2020-08-01 20:02:46,665   INFO  cfg.MODEL.BACKBONE_2D.NUM_FILTERS: [64, 128, 256]
2020-08-01 20:02:46,665   INFO  cfg.MODEL.BACKBONE_2D.UPSAMPLE_STRIDES: [1, 2, 4]
[... 150 lines cut ...]
        (0): ConvTranspose2d(256, 128, kernel_size=(4, 4), stride=(4, 4), bias=False)
        (1): BatchNorm2d(128, eps=0.001, momentum=0.01, affine=True, track_running_stats=True)
        (2): ReLU()
      )
    )
  )
  (dense_head): AnchorHeadSingle(
    (cls_loss_func): SigmoidFocalClassificationLoss()
[... cut ...]
2020-08-01 20:02:48,639   INFO  **********************Start training kitti_models/pointpillar(default)**********************
epochs:   0%|                                             | 0/1 [00:00<?, ?it/s]Traceback (most recent call last):
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 170, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/john/OpenPCDet/tools/train_utils/train_utils.py", line 69, in train_model
    with tqdm.trange(start_epoch, total_epochs, desc='epochs', dynamic_ncols=True, leave=(rank == 0)) as tbar:

sshaoshuai, 2020-08-01:
I never see this error before. Maybe it is related to the with statement of tqdm. 
You could try to re-install tqdm or refer to google for better understanding of this problem, just like https://stackoverflow.com/questions/51427729/python-error-attributeerror-enter.

Jsijon, 2020-08-02:
i reinstalled tqdm and it works . thanks

closed 2020-08-02 by Jsijon