open-mmlab/OpenPCDet #202: confused:where is the TWO VFE layers in SecondNet?

liyang0522, 2020-08-02:
the car mAP in the your  SecondNet is about 78.6, and the implements of the VFE is  meanVFE  , but the original paper Second has 2 VFE layers :VFE1(32) and VFE2(128) .
so my question is that when i replace the meanVFE with the two VFE layers , the mAP get worse! how does the  meanVFE get the  78.6 mAP better than the 2 VFE layers?  Thank you for your reply!!

sshaoshuai, 2020-08-03:
Currently the SECOND is the re-implemented of version 1.5, as presented in the original SECOND repo https://github.com/traveller59/second.pytorch. 

But I think the performance should drop with two extra MLP layers in the VFE.

liyang0522, 2020-08-03:
Thank you for your reply! 
If i have a new ieda that will be compared with the VoxelNet, but i use your SecondNet as backbone, mainly the sparseconv , instead of the original 3D CONV , so  can it be called the new method based on VoxelNet?

sshaoshuai, 2020-08-03:
I think it should be ok, since the basic idea of SECOND net also comes from the original VoxelNet. And you'd better introduce the improvement (i.e., sparse conv) compared with original VoxelNet.

liyang0522, 2020-08-04:
In MVX-Net: Multimodal VoxelNet for 3D Object Detection, the author dose not

closed 2020-08-04 by liyang0522