open-mmlab/OpenPCDet #204: Question about ROI-grid pooling

sakura-iv, 2020-08-03:
Hello, I am working on PV-RCNN model these days, and I print the net structure and find that:

> (roi_grid_pool_layer): StackSAModuleMSG(
>       (groupers): ModuleList(
>         (0): QueryAndGroup()
>         (1): QueryAndGroup()
>       )
>       (mlps): ModuleList(
>         (0): Sequential(
>           (0): Conv2d(131, 64, kernel_size=(1, 1), stride=(1, 1), bias=False)
>           (1): BatchNorm2d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
>           (2): ReLU()
>           (3): Conv2d(64, 64, kernel_size=(1, 1), stride=(1, 1), bias=False)
>           (4): BatchNorm2d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
>           (5): ReLU()
>         )
>         (1): Sequential(
>           (0): Conv2d(131, 64, kernel_size=(1, 1), stride=(1, 1), bias=False)
>           (1): BatchNorm2d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
>           (2): ReLU()
>           (3): Conv2d(64, 64, kernel_size=(1, 1), stride=(1, 1), bias=False)
>           (4): BatchNorm2d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
>           (5): ReLU()
>         )
>       )
>     )

It seems that there are not pooling operators in the net, so I wonder if you use convolution and BN operators to replace the pooling operators ? Or there are some other operators not showed here ?

Looking forward to your kindly reply, many thanks !

sshaoshuai, 2020-08-04:
Indeed, it doesn't have explicit pooling operator in RoI-grid pooling, since here the RoI-grid pooling means extract the specific features for each RoI, and it doesn't need specific pooling operator.

sakura-iv, 2020-08-04:
I read the code again, my understanding is that a customised QueryAndGroup operator was used to generate the grouped features from rois, and then you put them into max pooling operator to extract the final features. Am I right ? 
Many thanks !

sshaoshuai, 2020-08-06:
The max-pooling is used to extract the RoI-grid-wise features.