open-mmlab/OpenPCDet #205: Any model updates on waymo dataset since already won the waymo challenge

chowkamlee81, 2020-08-03:
@sshaoshuai 
Any model updates on waymo dataset since already won the waymo challenge

sshaoshuai, 2020-08-04:
Hi,  as you could see that we have released other dataset and SoTA models like KITTI and NuScenes, and there are many other things to do and we still need some time to sort out the waymo related models and configs. 
Please stay tuned for new updates!

chowkamlee81, 2020-08-04:
@sshaoshuai 
Algorithms trained on KITTI, not able to generalize well for our velodyne-64 data since kitti data is only few.
Nuscenes trained algos didn't perform well in real-time detection because of many false positives and not able to generalize well for our velodyne64 data eventhough dataset is huge.
Since waymo dataset is huge, a hope is built on as waymo uses 64Channel data. That's the reason we are looking for 64Channel lidar model wrt waymo dataset.
We don't have enough GPUs to train on waymo/Nuscenes, hence we are relying on your OpenPCDet since OpenPCDet releases models and which will be helpful for us.
Hope models will get released soon on Waymo dataset

closed 2020-08-06 by sshaoshuai