open-mmlab/OpenPCDet #208: PV-RCNN inference time too long to achieve real-time detection

Chunchunyoyo, 2020-08-04:
Hi, 

I inferenced PV-RCNN on TITAN Xp with your default config on kitti datasets, with batch size 1 and the inference time is about 130 ms. 

I analyze the time cost: 20ms for backbone 3d, 76 ms for pfe module and 27 ms for roi head. 

Any suggestions on shrinking the inference time?

Best regards.

sshaoshuai, 2020-08-04:
The most time consuming operation is may be the ball query. But it is not so easy to speed up this operation with CUDA codes. 
Also, you could set the `GRID_SIZE=4`, which could also speed up the training and inference time with a bit drop on the performance.

closed 2020-08-05 by sshaoshuai

Chunchunyoyo, 2020-08-05:
Thank you for your suggestion. In order to speed up the ball query operation, I think it can be achieved by searching neighbour points in the 3d nearby region in ball query module. 

In detail, using a hash table to store the valid points coordinates (which can get from SpcovTensor.indices), perform O(1) query for each keypoints to check whether there are some valid voxels in the neighbourhood. In this way, O(n) query can be avoid.

What do you think about this idea?

sshaoshuai, 2020-08-06:
Yes, we have tried that simple strategy and it could speed up the ball query operation slightly, and you could have a try.

Rzlnmc, 2022-07-14:
> Hi,
> 
> I inferenced PV-RCNN on TITAN Xp with your default config on kitti datasets, with batch size 1 and the inference time is about 130 ms.
> 
> I analyze the time cost: 20ms for backbone 3d, 76 ms for pfe module and 27 ms for roi head.
> 
> Any suggestions on shrinking the inference time?
> 
> Best regards.

Hi, if it is convenient, can you share the method of calculating the inference time? 
Thanks.