open-mmlab/OpenPCDet #209: Bug in sample_points function in data_processor.py when training PointRCNN_iou

ltphy, 2020-08-04:
In the following code:
```
       if (num_points < len(points)): 
            pts_depth = np.linalg.norm(points[:, 0:3], axis=1)
            pts_near_flag = pts_depth < 40.0
            far_idxs_choice = np.where(pts_near_flag==0)[0]
            near_idxs = np.where(pts_near_flag ==1)[0]
            near_idxs_choice = np.random.choice(near_idxs, num_points - len(far_idxs_choice), replace=False)
            choice = np.concatenate((near_idxs_choice, far_idxs_choice), axis=0 ) \
                if(len(far_idxs_choice) >0) else near_idxs_choice
            np.random.shuffle(choice)
```
I encounter this error when training point_rcnn_iou:
```
Near_idxs_choice = np.random.choice (near_idxs, num_points - len(far_idxs_choice), replace = False) 
 File "mtrand.pyx", line 951, in numpy.random.mtrand.RandomState.choice
ueError: negative dimensions are not allowed
```
This should be possibly because the length of `far_idxs_choice` is larger than the default number of points`16384`.

sshaoshuai, 2020-08-04:
Thank you for the report! 
It is indeed a bug and I didn't consider that situation since it is impossible in the KITTI dataset. 
You could submit a PR to fix it or I will find time to fix it.

ltphy, 2020-08-05:
Thank you very much. Btw, can I ask how can you choose the distance of points near flag to be smaller than 40 in this case. Is it based on the point cloud range?

sshaoshuai, 2020-08-05:
Yes, it is just a hyper-parameter based on the point cloud range. Actually I am not sure whether the performance will drop if we simply simple 16384 points from the whole point cloud. 
This parameter aimed to balance the number of points in nearby and far away areas since the random sampling may sample much more points in the nearby area.

ltphy, 2020-08-05:
I found that there are a few point cloud samples in KITTI dataset that the number of points outside the flag range (40) to be larger than 16384: 0561, 1152, 1194, 1251,.. . If this case happens, we should randomly choose 16384 points from the whole point cloud right?.

sshaoshuai, 2020-08-06:
I think this strategy should be fine.

closed 2020-08-06 by ltphy