open-mmlab/OpenPCDet #212: error when I run  python setup.py develop

Lyan1997, 2020-08-05:
when I run python setup.py develop, the following error occurred：

```
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
gcc -pthread -B /home/zyz/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/sdb1/zyan/lulu/lib/python3.7/site-packages/torch/include -I/home/sdb1/zyan/lulu/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/sdb1/zyan/lulu/lib/python3.7/site-packages/torch/include/TH -I/home/sdb1/zyan/lulu/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda:/include -I/home/zyz/anaconda3/include/python3.7m -c pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp -o build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:12:18: fatal error: cuda.h: 没有那个文件或目录
compilation terminated.
error: command 'gcc' failed with exit status 1
```

 How can I solve it？Please help me,thank you very much!!!

closed 2020-08-05 by Lyan1997

Jane-QinJ, 2021-12-09:
Have you solve this problem? I've met the same problem, I would appreciate it if you can help me!

Jane-QinJ, 2021-12-09:
Thanks， I've solved it! https://github.com/open-mmlab/OpenPCDet/issues/572