open-mmlab/OpenPCDet #213: demo works in v0.2 but no in v0.3

Shawn-Zhuang, 2020-08-05:
I have successfully run the demo.py in v0.2. 
Then I change the name of directory "OpenPCDet" to "OpenPCDet_v0.2" to avoid clash when I git clone the new version of OpenPCDet, and try the demo.py again. However, this time I met such  errors:
```
2020-08-05 21:54:07,891   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2020-08-05 21:54:07,942   INFO  Total number of samples: 	1
Segmentation fault (core dumped)
```
which occurs when it processes
`model = build_network(model_cfg=cfg.MODEL, num_class=len(cfg.CLASS_NAMES), dataset=demo_dataset`)

I see some "Segmentation fault (core dumped)" errors happening when the compilation went wrong.

The following is the log of the compilation, which include a lot of warnings but in the end succeed:

```
(base) shawn@shawn-HP:~/OpenPCDet$ python setup.py develop
running develop
running egg_info
creating pcdet.egg-info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
writing manifest file 'pcdet.egg-info/SOURCES.txt'
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
creating build
creating build/temp.linux-x86_64-3.7
creating build/temp.linux-x86_64-3.7/pcdet
creating build/temp.linux-x86_64-3.7/pcdet/ops
creating build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms
creating build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src
gcc -pthread -B /home/shawn/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/TH -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/shawn/anaconda3/include/python3.7m -c pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp -o build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
gcc -pthread -B /home/shawn/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/TH -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/shawn/anaconda3/include/python3.7m -c pcdet/ops/iou3d_nms/src/iou3d_nms_api.cpp -o build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
gcc -pthread -B /home/shawn/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/TH -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/shawn/anaconda3/include/python3.7m -c pcdet/ops/iou3d_nms/src/iou3d_nms.cpp -o build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
/usr/local/cuda/bin/nvcc -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/TH -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/shawn/anaconda3/include/python3.7m -c pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --compiler-options '-fPIC' -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
creating build/lib.linux-x86_64-3.7
creating build/lib.linux-x86_64-3.7/pcdet
creating build/lib.linux-x86_64-3.7/pcdet/ops
creating build/lib.linux-x86_64-3.7/pcdet/ops/iou3d_nms
g++ -pthread -shared -B /home/shawn/anaconda3/compiler_compat -L/home/shawn/anaconda3/lib -Wl,-rpath=/home/shawn/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_cpu.o build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms.o build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -L/usr/local/cuda/lib64 -lcudart -o build/lib.linux-x86_64-3.7/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-37m-x86_64-linux-gnu.so
building 'pcdet.ops.roiaware_pool3d.roiaware_pool3d_cuda' extension
creating build/temp.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d
creating build/temp.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d/src
gcc -pthread -B /home/shawn/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/TH -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/shawn/anaconda3/include/python3.7m -c pcdet/ops/roiaware_pool3d/src/roiaware_pool3d.cpp -o build/temp.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d/src/roiaware_pool3d.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=roiaware_pool3d_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
/usr/local/cuda/bin/nvcc -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/TH -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/shawn/anaconda3/include/python3.7m -c pcdet/ops/roiaware_pool3d/src/roiaware_pool3d_kernel.cu -o build/temp.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d/src/roiaware_pool3d_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --compiler-options '-fPIC' -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=roiaware_pool3d_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
creating build/lib.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d
g++ -pthread -shared -B /home/shawn/anaconda3/compiler_compat -L/home/shawn/anaconda3/lib -Wl,-rpath=/home/shawn/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d/src/roiaware_pool3d.o build/temp.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d/src/roiaware_pool3d_kernel.o -L/usr/local/cuda/lib64 -lcudart -o build/lib.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-37m-x86_64-linux-gnu.so
building 'pcdet.ops.roipoint_pool3d.roipoint_pool3d_cuda' extension
creating build/temp.linux-x86_64-3.7/pcdet/ops/roipoint_pool3d
creating build/temp.linux-x86_64-3.7/pcdet/ops/roipoint_pool3d/src
gcc -pthread -B /home/shawn/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/TH -I/home/shawn/anaconda3/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/inclu [... cut ...]
Adding setuptools 45.2.0.post20200210 to easy-install.pth file
Installing easy_install script to /home/shawn/anaconda3/bin

Using /home/shawn/anaconda3/lib/python3.7/site-packages
Searching for llvmlite==0.31.0
Best match: llvmlite 0.31.0
Adding llvmlite 0.31.0 to easy-install.pth file

Using /home/shawn/anaconda3/lib/python3.7/site-packages
Finished processing dependencies for pcdet==0.3.0+96a6092
```
Could it be possible that I should remove the original version of OpenPCDet directory and remove the site-package in the conda base environment?

sshaoshuai, 2020-08-06:
You'd better try to install it in a clean directory of `OpenPCDet`, since your build log seems there are some history files within `OpenPCDet`.

Shawn-Zhuang, 2020-08-07:
@sshaoshuai when I comment the following lines in the demo.py, everything is OK, I can print the predicts.
```
# import mayavi.mlab as mlab
# mlab.show(stop=False)
```
But I can not visualize the points if I comment them.
My mayavi version is 4.7.1, is the version too old? But this version works in OPenPCDet v0.2.
I try to install the v0.3 on a new Ubuntu system, there occurs the same problem.

Shawn-Zhuang, 2020-08-07:
I know how to fix this. Before I install mayavi, I have PyQt5 installed. However, Mayavi uses PyQt4. After I remove PtQt5 the visualization works.

closed 2020-08-07 by Shawn-Zhuang