open-mmlab/OpenPCDet #214: SAMPLE_GROUPS

hcyoo93, 2020-08-05:
Hello @sshaoshuai,

Thank you for your great work!

I am wondering "SAMPLE_GROUPS" from nuscenes_dataset.yaml

```
SAMPLE_GROUPS: [
              'car:2','truck:3', 'construction_vehicle:7', 'bus:4', 'trailer:6',
              'barrier:2', 'motorcycle:6', 'bicycle:6', 'pedestrian:2', 'traffic_cone:2']
```
       
However, I found that your comment
"Class-balanced sampling of nuScenes dataset from https://arxiv.org/abs/1908.09492"  

and the paper proposes

```
SAMPLE_GROUPS: [
              'car:2','truck:3', 'construction_vehicle:6', 'bus:7', 'trailer:4',
              'barrier:6', 'motorcycle:2', 'bicycle:6', 'pedestrian:2', 'traffic_cone:2']
```
which is different from the configuration.

I am curious that this difference is the intentional modification.

Thank you!

sshaoshuai, 2020-08-06:
No, it is not intentional modification. And I just follow the config of [here](https://github.com/poodarchu/Det3D/blob/master/examples/cbgs/configs/nusc_all_vfev3_spmiddleresnetfhd_rpn2_mghead_syncbn.py#L213-L228).

closed 2020-08-06 by sshaoshuai

hcyoo93, 2020-08-11:
Thanks for the reply!