open-mmlab/OpenPCDet #216: New requirements in 0.3.0?

jacoblambert, 2020-08-06:
I was able to install OpenPCDET 0.3.0 fine but when trying to test the nuscenes model I get
`/ops/pointnet2/pointnet2_stack/pointnet2_stack_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _Z36three_interpolate_grad_wrapper_stackN2at6TensorES0_S0_S0_
`
I'm using Cuda 10, PyTorch 1.1, SpConv 1.0, Python 3.6.5. Are these still adequate for 0.3.0?
I tried starting again from a clean conda environment with these specs but no luck. Any idea what the issue might be?

sshaoshuai, 2020-08-06:
You need to delete the `OpenPCDet/build` directory before you run `python setup.py develop`, have you tried it?

jacoblambert, 2020-08-06:
Thank you!! Solved.

closed 2020-08-06 by jacoblambert

clw5180, 2024-05-12:
> You need to delete the `OpenPCDet/build` directory before you run `python setup.py develop`, have you tried it?

大佬tql，一眼看出问题

thelandord, 2024-06-13:
牛呗！