open-mmlab/OpenPCDet #217: pointnet2_batch_cuda.cpython-37m-x86_64-linux-gnu.so: undefined symbol

Kemo-Huang, 2020-08-06:
It seems that I cannot correctly build pointnet2 but there is no warning during `python setup.py develop`. It is incompatible with the newest pytorch?

Environment:
- Ubuntu 18.04
- Python 3.7.7
- PyTorch 1.6.0
- CUDA 10.2 (installed by `apt install cuda-toolkit-10-2`)

Command:
`python test.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 2 --ckpt pv_rcnn_8369.pth`

Error message:
- ImportError: ~/Github/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/pointnet2_batch_cuda.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZN3c105ErrorC1ENS_14SourceLocationERKSs

jacoblambert, 2020-08-06:
Maybe same problem as me: https://github.com/open-mmlab/OpenPCDet/issues/216

sshaoshuai, 2020-08-07:
Please re-build it as mentioned here https://github.com/open-mmlab/OpenPCDet/issues/216.

closed 2020-08-09 by Kemo-Huang