open-mmlab/OpenPCDet #221: Nuscenes trained models performs poorly but kitti pretrained relaesed models generallize well on custom pointclouds

chowkamlee81, 2020-08-07:
Even though nuscnes trained models has been trained well , models are not able to generalize well on custom lidar pointclouds whereas kitti relesaed pretrained models performs well.

Strange..

Kitti pretrained models only work for forward view on pointcloud data? or it works only for entire 360 degree?

sshaoshuai, 2020-08-07:
NuScenes uses 10 sweeps to train the models and the lidar only has 32 beams, so it is reasonable the performance is bad when generalizing to new point clouds.

chowkamlee81, 2020-08-07:
Training with sweeps=1 on nuscenes, generalize well for other pointcloud sensor data?

When PVRCNN model for Nuscenes will get released?

sshaoshuai, 2020-08-09:
Sorry, I just found a bug about loading nuscene pretrained models, please refer to https://github.com/open-mmlab/OpenPCDet/issues/225. 

Transferring PV-RCNN to NuScenes needs carefully tune the parameters, and you could have a try.

sshaoshuai, 2020-08-10:
The provided NuScenes models have been updated, please re-download it if you need it.

closed 2020-08-10 by sshaoshuai

kadebudongtan, 2022-03-02:
> 

So nuscenes models can't be generalized, what's the point of their existence?