open-mmlab/OpenPCDet #222: error when I run 'python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml

Lyan1997, 2020-08-07:
I want to only  input point cloud. So I did the following：

After I deleted the codes about image and calib in the kitti_dataset.py, then when  I run 'python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml', the following error occurred:

```
File "/home/zyz/anaconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/zyz/anaconda3/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/sdb1/zyan/lulu/OpenPCDet-2.0/OpenPCDet-2.0.0/pcdet/datasets/kitti/kitti_dataset.py", line 438, in <module>
    save_path=ROOT_DIR / 'data' / 'kitti'
  File "/home/sdb1/zyan/lulu/OpenPCDet-2.0/OpenPCDet-2.0.0/pcdet/datasets/kitti/kitti_dataset.py", line 398, in create_kitti_infos
    kitti_infos_train = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
  File "/home/sdb1/zyan/lulu/OpenPCDet-2.0/OpenPCDet-2.0.0/pcdet/datasets/kitti/kitti_dataset.py", line 191, in get_infos
    return list(infos)
  File "/home/zyz/anaconda3/lib/python3.7/concurrent/futures/_base.py", line 598, in result_iterator
    yield fs.pop().result()
  File "/home/zyz/anaconda3/lib/python3.7/concurrent/futures/_base.py", line 428, in result
    return self.__get_result()
  File "/home/zyz/anaconda3/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/home/zyz/anaconda3/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/sdb1/zyan/lulu/OpenPCDet-2.0/OpenPCDet-2.0.0/pcdet/datasets/kitti/kitti_dataset.py", line 178, in process_single_scene
    corners_lidar = box_utils.boxes_to_corners_3d(gt_boxes_lidar)
  File "/home/sdb1/zyan/lulu/OpenPCDet-2.0/OpenPCDet-2.0.0/pcdet/utils/box_utils.py", line 47, in boxes_to_corners_3d
    corners3d = common_utils.rotate_points_along_z(corners3d.view(-1, 8, 3), boxes3d[:, 6]).view(-1, 8, 3)
IndexError: index 6 is out of bounds for dimension 1 with size 4
```
Please help me solve it! Thank you so much!!!

NLCharles, 2020-11-06:
You are trying a hard way. The easiest way to go is to set your data format to kitti-like, including changing calibration matrices to identity matrices, and setting proper paths. If you delete part of the codes, you will ruin the dataset builder.
You may also reimplement the whole dataset function, which I just did recently.

Lyan1997, 2020-11-06:
> You are trying a hard way. The easiest way to go is to set your data format to kitti-like, including changing calibration matrices to identity matrices, and setting proper paths. If you delete part [...]

But I have learned that this model could just need the LiDAR datasets through reading his papers. Therefore, I want to know how to train this model through using the LiDAR data I only have now.

NLCharles, 2020-11-06:
> But I have learned that this model could just need the LiDAR datasets through reading his papers. Therefore, I want to know how to train this model through using the LiDAR data I only have now.

You are right about one thing: the model DOES predict bounding box from lidar data only. However, the kitti dataset input pipeline wasn't meant for a non-image dataset -- in many KittiDataset methods, the calib file and image information is required and is processed. 
To avoid this, you should either overide them or make your dataset looks like a kitti dataset, if you would like to use kitti configs and methods(and this still requires some alters in the dataset function). I did this a lot when I was a student.
The author explained in the dataset.py that if you want another approach, thatis, to have the model trained and tested on your own dataset, you will have to reimplement several methods that meet the minimum requirement for the base class(requiring only lidar data and annotations). And actually you will have to pay attention to more than described by the author.

Lyan1997, 2020-11-06:
> You are right about one thing: the model DOES predict bounding box from lidar data only. However, the kitti dataset input pipeline wasn't meant for a non-image dataset -- in many KittiDataset [...]

I'm sorry I do not know how to overide the kitti dataset.

closed 2022-03-08 by github-actions[bot]