open-mmlab/OpenPCDet #223: CUDA kernel failed

LeoAnnArbor, 2020-08-07:
Hi, 

Thanks for sharing the repo. I was trying to replicate the PV-RCNN work. But after building the ws and sparse conv. I always get CUDA kernel error. Following are the outputs:

from **demo.py**
2020-08-07 15:30:59,074   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2020-08-07 15:30:59,098   INFO  Total number of samples: 	1
2020-08-07 15:31:00,720   INFO  -----------------build_network done-------------------------
2020-08-07 15:31:00,721   INFO  ==> Loading parameters from checkpoint pv_rcnn_8369.pth to CPU
2020-08-07 15:31:01,218   INFO  ==> Done (loaded 367/367)
2020-08-07 15:31:01,236   INFO  Visualized sample index: 	1
**CUDA kernel failed : invalid device function
Segmentation fault (core dumped)**

from **test.py**
2020-08-07 15:33:42,750   INFO  **********************Start logging**********************
2020-08-07 15:33:42,751   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-08-07 15:33:42,751   INFO  cfg_file         cfgs/kitti_models/pv_rcnn.yaml
2020-08-07 15:33:42,751   INFO  batch_size       8
2020-08-07 15:33:42,751   INFO  workers          4
2020-08-07 15:33:42,751   INFO  extra_tag        default
2020-08-07 15:33:42,751   INFO  ckpt             pv_rcnn_8369.pth
2020-08-07 15:33:42,751   INFO  launcher         none
2020-08-07 15:33:42,751   INFO  tcp_port         18888
2020-08-07 15:33:42,751   INFO  local_rank       0
2020-08-07 15:33:42,751   INFO  set_cfgs         None
2020-08-07 15:33:42,751   INFO  max_waiting_mins 30
2020-08-07 15:33:42,751   INFO  start_epoch      0
2020-08-07 15:33:42,751   INFO  eval_tag         default
2020-08-07 15:33:42,752   INFO  eval_all         False
2020-08-07 15:33:42,752   INFO  ckpt_dir         None
2020-08-07 15:33:42,752   INFO  save_to_file     False
2020-08-07 15:33:42,752   INFO  cfg.ROOT_DIR: /media/zhen/New_Volume/shenlan/Point_Cloud_Processing/L6/OpenPCDet
2020-08-07 15:33:42,752   INFO  cfg.LOCAL_RANK: 0
2020-08-07 15:33:42,752   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-08-07 15:33:42,752   INFO  
cfg.DATA_CONFIG = edict()
2020-08-07 15:33:42,752   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-08-07 15:33:42,752   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2020-08-07 15:33:42,752   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2020-08-07 15:33:42,752   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-08-07 15:33:42,752   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-08-07 15:33:42,752   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-08-07 15:33:42,752   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2020-08-07 15:33:42,752   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2020-08-07 15:33:42,752   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2020-08-07 15:33:42,752   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-08-07 15:33:42,752   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-08-07 15:33:42,753   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-08-07 15:33:42,753   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-08-07 15:33:42,753   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-08-07 15:33:42,753   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-08-07 15:33:42,753   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2020-08-07 15:33:42,753   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2020-08-07 15:33:42,753   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2020-08-07 15:33:42,753   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2020-08-07 15:33:42,753   INFO  
cfg.MODEL = edict()
2020-08-07 15:33:42,753   INFO  cfg.MODEL.NAME: PVRCNN
2020-08-07 15:33:42,753   INFO  
cfg.MODEL.VFE = edict()
2020-08-07 15:33:42,753   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2020-08-07 15:33:42,753   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2020-08-07 15:33:42,753   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2020-08-07 15:33:42,753   INFO  
[... 194 lines cut ...]
NVIDIA Driver Version: 430.50       
CUDA Version: 10.1  

Selected packages:

# Name                    Version                 Build                 Channel
_libgcc_mutex          0.1                       main  
libgcc-ng                   9.1.0                    hdf63c60_0  
libstdcxx-ng              9.1.0                    hdf63c60_0  
mayavi                      4.7.2.dev0           pypi_0                   pypi
pcdet                         0.3.0+96a6092   dev_0                    <develop>
pickleshare               0.7.5                    pypi_0                  pypi
pip                            20.1.1                  py36_1  
pyqt5                        5.15.0                  pypi_0                  pypi
pyqt5-sip                  12.8.0                  pypi_0                  pypi
python                      3.6.10                  h7579374_2  
python-dateutil         2.8.1                    pypi_0                  pypi
pytorch                     1.6.0           py3.6_cuda10.1.243_cudnn7.6.3_0    pytorch
spconv                     1.2.1                    pypi_0                  pypi
torch                         1.6.0                    pypi_0                 pypi
torchvision                0.7.0                    py36_cu101        pytorch
vtk                            9.0.1                    pypi_0                 pypi
wheel                       0.34.2                   py36_0  

Any suggestions would be really appreciated! Thanks!

thptai, 2020-08-08:
Try to install cuda 10.2

I had the same problem yesterday (with cuda 10.0)

I updated my cuda to 10.2 and reinstall everything and it works well now.

sshaoshuai, 2020-08-14:
Feel free to re-open it if you still have problems.

closed 2020-08-14 by sshaoshuai

matteo-bastico, 2020-09-24:
I have the same problem even with CUDA 10.2. Any suggestion?

MartinHahner88, 2020-09-24:
use CUDA 9.0 if possible