open-mmlab/OpenPCDet #225: Cannot reproduce the NuScenes result with the provided model

pyun-ram, 2020-08-08:
Hi @sshaoshuai,

Thank you for the awesome work of providing supports to the NuScenes dataset!
I tried to use the SECOND(CBGS) model with the weights you provided in the model zoo, and evaluated it on the official NuScenes dataset v1.0-trainval.
The result I got is far from the reported results in the Model Zoo. Could you please give me some help?

The data I used was downloaded from https://www.nuscenes.org/download, 
including the v1.0-trainval_meta.tar, v1.0-trainval[01-10]_blobs_lidar.tar, v1.0-trainval[01-10]_keyframes.tar.
I uncompressed them and process the data with
```
python -m pcdet.datasets.nuscenes.nuscenes_dataset --func create_nuscenes_infos \
    --cfg_file tools/cfgs/dataset_configs/nuscenes_dataset.yaml \
    --version v1.0-trainval
```

The weight I used is downloaded from the model zoo: https://drive.google.com/file/d/1s34D8g-h65qDyoYbgCraxcZQwinbxhaY/view?usp=sharing
I conducted the evaluation with
```
python test.py --cfg_file cfgs/nuscenes_models/cbgs_second_multihead.yaml --batch_size 10 --ckpt ../cbgs_second_multihead_nds6229.pth
```

**The final evaluation mAP is 16.5 far from the reported result: 50.59.**
I attached the evaluation log file for your reference.
[log_eval_20200808-064040.txt](https://github.com/open-mmlab/OpenPCDet/files/5044810/log_eval_20200808-064040.txt)

I am using the docker with python3.7 and **PyTorch 1.4 spconv1.2**.
I did not encounter the mentioned spconv1.2 problem related to "nvcc fatal: unknown -Wall" as the README of https://github.com/traveller59/spconv.

muzi2045, 2020-08-08:
with the test of release pretrained model trained in Nuscenes dataset, it's result are even worse than trained on KITTI, that's really wired.

sshaoshuai, 2020-08-09:
Sorry about that, I just found a small bug to load nuscene pretrained models. I will fix it soon. 
Currently you could modify the following line to load the pretrained models: 

https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/models/backbones_2d/base_bev_backbone.py#L49

Modify `if stride >= 1:` to `if stride > 1`.

pyun-ram, 2020-08-10:
Thanks for your help! 
By checking the log file, I found I did not load the parameter "backbone_2d.deblocks.0.0.weight" sucessfully.
After mannuly transposing the weight, I made it and get the exactly second(CBGS) result as you reported.
Thank you so much!

closed 2020-08-10 by pyun-ram