open-mmlab/OpenPCDet #226: [BUG] empty np.ndarray is not None

Gltina, 2020-08-08:
Hi,

## Description

When pressing my own data as input to test 'demo.py', I found an extremely weird thing that some data could be shown but others cannot. Finally, I found that it was caused by the empty detected target, as shown below:

[image]

where `ref_label` is empty(`[]`), but it still goes into the `else` codeblock to `print('ref_label is not None')`, which is shown on the left in the image above.

**For verifying this is a bug, I reproduced this again as below:**

// test on windows with python3.6
``` python
Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy as np
>>> a=np.array([])
>>> a is None
False
>>> len(a)
0
```

## How to fix it to show visualize any point cloud data

For show point cloud that has no detected target, you can using `.size` or `len()` to be new if-statement instead of 'np.ndarray is None', as following:

``` python
if len(ref_labels) > 0:
   ...
else:
   ...
```
[image]

What's more, in `demo.py`, I found there are many codes using `is None` as if-statement.

## Others

I guess this might be the reason for the problem mentioned in [issue](https://github.com/open-mmlab/OpenPCDet/issues/176#issuecomment-668022490) @kamillasak

Thanks,

sshaoshuai, 2020-08-10:
Hi, 
Thank you for the detailed bug description, I have fixed in https://github.com/open-mmlab/OpenPCDet/issues/226. 
You could check it now.

closed 2020-08-10 by sshaoshuai

Petros626, 2024-10-01:
@sshaoshuai your link is this issue, I guess you wanted to mention another issue with the solved problem?