open-mmlab/OpenPCDet #228: question about model export to onnx

chasingw, 2020-08-10:
I want to use TensorRT to delpoy the model. and I try to export the model to onnx, but I found this error when I use `export_onnx` function 
```
RuntimeError: Only tuples, lists and Variables supported as JIT inputs/outputs. Dictionaries and strings are also accepted but their usage is not recommended. But got unsupported type numpy.ndarray
```
and this seems to be because forward function use dict as input param. can I have some non-destructive methods to support export to onnx?

sshaoshuai, 2020-08-10:
I am not familiar with onnx, so you may need to transform the dict input to specific variables one by one.

muzi2045, 2020-08-19:
pytorch don't support dict when export model to onnx.
so here is something to change in networek input and output, the dict input --> list input
here is there part onnx export from OpenPCDet codebase:

[image]

[image]

[image]

chasingw, 2020-08-27:
@sshaoshuai @muzi2045 thanks for your reply. I found it can support dict type since pytorch 1.3 [#13040](https://github.com/pytorch/pytorch/issues/13040),the error above is because the dict contain numpy.ndarray type, so I remove it and I can export to onnx but the onnx model seems more complicated and contain some ATen operators which can not be run by tensorRT runtime.which model you export to onnx, and could you tell me some detail, and if I want to use different model, do I need to modify one by one, thanks

muzi2045, 2020-08-28:
I convert the original pointpillars model without multi-head.
and using onnx-simpilfy tool to remove those onnx extra ops in export process.

NLCharles, 2020-11-06:
> I convert the original pointpillars model without multi-head.
> and using onnx-simpilfy tool to remove those onnx extra ops in export process.

Is it possible to transplant this project to tensorrt? Farthest Point Sampling, Feature Propagation and Sparse Convolution are quite challenging to me.

tzhong518, 2020-11-09:
> > I convert the original pointpillars model without multi-head.
> > and using onnx-simpilfy tool to remove those onnx extra ops in export process.
> 
> Is it possible to transplant this project to tensorrt? Farthest Point Sampling, Feature Propagation and Sparse Convolution are quite challenging to me.

Hi, I want to export the model containing sparse convolution to tensorrt, too. 
Have you found any clue about how to do this?

NLCharles, 2020-11-09:
> Hi, I want to export the model containing sparse convolution to tensorrt, too.
> Have you found any clue about how to do this?
My team are assessing relevant models. Some of the custom ops are not GPU-friendly and may be hard to optimize. If we can't boost the model to 10fps, then you know there's no point to do the transplantation.

tzhong518, 2020-11-09:
> My team are assessing relevant models. Some of the custom ops are not GPU-friendly and may be hard to optimize. If we can't boost the model to 10fps, then you know there's no point to do the [...]

Thank you for the information. 
I only tested PartA2 model in pytorch and it's not far from 10fps. I thought transplanting it to tensorrt could boost its speed.

Could you please tell me which ops are not GPU-friendly, e.g sparse convolution? 

Looking forward to your results.

NLCharles, 2020-11-09:
> Thank you for the information.
> I only tested PartA2 model in pytorch and it's not far from 10fps. I thought transplanting it to tensorrt could boost its speed.
> 
> Could you please tell me which ops are not GPU-friendly, e.g sparse convolution?
> 
> Looking forward to your results.

Thank you for your reply. Did your run time include preprocess and post process? 

Ops utilized by PointNet++ modules, set abstraction and forward interpolation, involve intensivelogical oprations. 

For voxelnet, voxelization of the input costs a lot of time, and sparse convolution must be wrapped into tensorrt plugin.

tzhong518, 2020-11-09:
> Thank you for your reply. Did your run time include preprocess and post process?
>
> Ops utilized by PointNet++ modules, set abstraction and forward interpolation, involve intensivelogical oprations. [...]

Oh my fault, it didn't include preprocess and post process time.
I'm looking into voxelnet for faster speed.

And sorry, I'm a newbie to tensorrt. Do you mean wrapping sparse convolution into tensorrt plugin when using torch2rt?
But it seems that torch2rt doesn't support dynamic shape.

NLCharles, 2020-11-09:
> 
> Oh my fault, it didn't include preprocess and post process time.
> I'm looking into voxelnet for faster speed.
> 
> And sorry, I'm a newbie to tensorrt. Do you mean wrapping sparse convolution into tensorrt plugin when using torch2trt?
> But it seems that torch2trt doesn't support dynamic shape.
>

Model conversion is a big cost for maintanance of a model with custom ops. For example, by using TFTRT you will maintain 2 versions of your model, TF ver and pb ver (and torchrt too), and you will ultimately implement and wrap some custom ops that are not supported by TensorRT.

Besides, for non gpu ops between gpu ops, data will be copied and moved between local memory  and gpu device memory, causing hugh IO cost. 

Our leader decides further model deployment work be done based purely on TensorRT.

wolflion521, 2021-01-18:
> pytorch don't support dict when export model to onnx.
> so here is something to change in networek input and output, the dict input --> list input
> here is three part onnx export from OpenPCDet codebase:
> 
> [image]
> 
> [image]
> 
> [image]
Hi , the three images are no longer available, can you paste them again please? Thanks a lot.

AngYikHong, 2021-03-09:
> @sshaoshuai @muzi2045 thanks for your reply. I found it seems to support dict type since pytorch 1.3 [#13040](https://github.com/pytorch/pytorch/issues/13040),the error above is because the dict contain numpy.ndarray type, so I remove it and I can export to onnx but the onnx model seems more complicated and contain some ATen operators which can not be run by tensorRT runtime.which model you export to onnx, and could you tell me some detail, and if I want to use different model, do I need to modify one by one, thanks

@chasingw Hi, I have the same issue with u, may I know how u set the input argument in torch.onnx.export?

Mfertega, 2021-03-30:
Hi @muzi2045, I have been trying to compile Poinpillars into 3 .onnx files but I'm running into some warnings in some transformations, did you modify the base code of the net to make those operations before forwarding? 
I'm new to exporting in onnx format so if you can share your steps to transform the net will be very helpfull.

```
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/backbones_3d/vfe/pillar_vfe.py:104: TracerWarning: There are 2 live references to the data region being modified when tracing in-place operator copy_ (possibly due to an assignment). This might cause the trace to be incorrect, because all other views that also reference this data will not reflect this change in the trace! On the other hand, if all other views use the same memory chunk, but are disjoint (e.g. are outputs of torch.split), this might still be safe.
  f_center[:, :, 0] = voxel_features[:, :, 0] - (coords[:, 3].to(voxel_features.dtype).unsqueeze(1) * self.voxel_x + self.x_offset)
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/backbones_3d/vfe/pillar_vfe.py:105: TracerWarning: There are 2 live references to the data region being modified when tracing in-place operator copy_ (possibly due to an assignment). This might cause the trace to be incorrect, because all other views that also reference this data will not reflect this change in the trace! On the other hand, if all other views use the same memory chunk, but are disjoint (e.g. are outputs of torch.split), this might still be safe.
  f_center[:, :, 1] = voxel_features[:, :, 1] - (coords[:, 2].to(voxel_features.dtype).unsqueeze(1) * self.voxel_y + self.y_offset)
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/backbones_3d/vfe/pillar_vfe.py:106: TracerWarning: There are 2 live references to the data region being modified when tracing in-place operator copy_ (possibly due to an assignment). This might cause the trace to be incorrect, because all other views that also reference this data will not reflect this change in the trace! On the other hand, if all other views use the same memory chunk, but are disjoint (e.g. are outputs of torch.split), this might still be safe.
  f_center[:, :, 2] = voxel_features[:, :, 2] - (coords[:, 1].to(voxel_features.dtype).unsqueeze(1) * self.voxel_z + self.z_offset)
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/backbones_3d/vfe/pillar_vfe.py:30: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  if inputs.shape[0] > self.part:
/home/miguel/PycharmProjects/OpenPCDet_upgraded/venv/lib/python3.6/site-packages/torch/onnx/utils.py:172: UserWarning: We detected that you are modifying a dictionnary that is an input to your model. Note that dictionaries are allowed as inputs in ONNX but they should be handled with care. Usages of dictionaries is not recommended, and should not be used except for configuration use. Also note that the order and values of the keys must remain the same. 
  warnings.warn(warning)
/home/miguel/PycharmProjects/OpenPCDet_upgraded/venv/lib/python3.6/site-packages/torch/onnx/symbolic_opset9.py:1283: UserWarning: ONNX export unsqueeze with negative axis -1 might cause the onnx model to be incorrect. Negative axis is not supported in ONNX. Axis is converted to 2 based on input shape at export time. Passing an tensor of different rank in execution will be incorrect.
  "Passing an tensor of different rank in execution will be incorrect.")
0 cost time: 0.043920278549194336
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/backbones_2d/map_to_bev/pointpillar_scatter.py:18: TracerWarning: Converting a tensor to a Python number might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  batch_size = coords[:, 0].max().int().item() + 1
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/backbones_2d/map_to_bev/pointpillar_scatter.py:32: TracerWarning: There are 2 live references to the data region being modified when tracing in-place operator index_put_. This might cause the trace to be incorrect, because all other views that also reference this data will not reflect this change in the trace! On the other hand, if all other views use the same memory chunk, but are disjoint (e.g. are outputs of torch.split), this might still be safe.
  spatial_feature[:, indices] = pillars
1 cost time: 0.15864205360412598
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/backbones_2d/base_bev_backbone.py:95: TracerWarning: Converting a tensor to a Python integer might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  stride = int(spatial_features.shape[2] / x.shape[2])
2 cost time: 0.19463157653808594
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/dense_heads/anchor_head_template.py:265: TracerWarning: There are 2 live references to the data region being modified when tracing in-place operator copy_ (possibly due to an assignment). This might cause [... cut ...]
  batch_box_preds[..., 6] = dir_rot + dir_offset + period * dir_labels.to(batch_box_preds.dtype)
/home/miguel/PycharmProjects/OpenPCDet_upgraded/OpenPCDet/pcdet/models/dense_heads/anchor_head_single.py:73: TracerWarning: torch.tensor results are registered as constants in the trace. You can safely ignore this warning if you use this function to create tensors out of constant variables that would be the same every time you call this function. In any other case, this might cause the trace to be incorrect.
  data_dict['cls_preds_normalized'] = torch.tensor(False)
```

shubham96jain, 2021-04-05:
> Hi @muzi2045, I have been trying to compile Poinpillars into 3 .onnx files but I'm running into some warnings in some transformations, did you modify the base code of the net to make those [...]

Hi @Mfertega and @muzi2045 , 

I am also in the process of converting pointpillars to ONNX. I am getting the below error:

`*** RuntimeError: Only tuples, lists and Variables supported as JIT inputs/outputs. Dictionaries and strings are also accepted but their usage is not recommended. But got unsupported type bool`

I am using PyTorch 1.3.1

Help is much appreciated!

Thanks

z-hou, 2021-05-17:
hi guys, I also convert the model to onnx, and I use pytorch v1.1.0, and I got "Aten" op in my onnx model which cannot be executed by onnx runtime.
 
[]([]([image]))

Did this error caused by wrong torch version? Hoping for your answer

wangy12, 2022-01-30:
Related questions and answers
https://github.com/open-mmlab/mmdetection3d/issues/203
https://github.com/speshowBUAA/mmdet3d_onnx_tools
https://github.com/CarkusL/CenterPoint

closed 2022-03-16 by github-actions[bot]

zhoujinhai, 2024-04-19:
> I convert the original pointpillars model without multi-head. and using onnx-simpilfy tool to remove those onnx extra ops in export process.

Can you provide any examples over there，Thanks！

zhoujinhai, 2024-04-23:
> @sshaoshuai @muzi2045 thanks for your reply. I found it seems to support dict type since pytorch 1.3 [#13040](https://github.com/pytorch/pytorch/issues/13040),the error above is because the dict [...]

Do you remember how to fix the bug?