open-mmlab/OpenPCDet #230: No local packages or working download links found for spconv,error: Could not find suitable distribution for Requirement.parse('spconv')

AravindG1508, 2020-08-10:
xyz@pvrcnn:~/OpenPCDet$ sudo python3 setup.py develop
/usr/local/lib/python3.7/dist-packages/setuptools/dist.py:458: UserWarning: Normalizing '0.3.0+0741976' to '0.3.0+741976'
  warnings.warn(tmpl.format(**locals()))
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
/home/xyz/.local/lib/python3.7/site-packages/torch/utils/cpp_extension.py:304: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
copying build/lib.linux-x86_64-3.7/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-37m-x86_64-linux-gnu.so -> pcdet/ops/iou3d_nms
copying build/lib.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-37m-x86_64-linux-gnu.so -> pcdet/ops/roiaware_pool3d
copying build/lib.linux-x86_64-3.7/pcdet/ops/roipoint_pool3d/roipoint_pool3d_cuda.cpython-37m-x86_64-linux-gnu.so -> pcdet/ops/roipoint_pool3d
copying build/lib.linux-x86_64-3.7/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_stack_cuda.cpython-37m-x86_64-linux-gnu.so -> pcdet/ops/pointnet2/pointnet2_stack
copying build/lib.linux-x86_64-3.7/pcdet/ops/pointnet2/pointnet2_batch/pointnet2_batch_cuda.cpython-37m-x86_64-linux-gnu.so -> pcdet/ops/pointnet2/pointnet2_batch
Creating /usr/local/lib/python3.7/dist-packages/pcdet.egg-link (link to .)
pcdet 0.3.0+741976 is already the active version in easy-install.pth

Installed /home/xyz/OpenPCDet
Processing dependencies for pcdet==0.3.0+741976
Searching for spconv
Reading https://pypi.org/simple/spconv/
Couldn't find index page for 'spconv' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.org/simple/
No local packages or working download links found for spconv
error: Could not find suitable distribution for Requirement.parse('spconv')

jihanyang, 2020-08-11:
Please check https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md to install spconv.

AravindG1508, 2020-08-11:
thanks for reply. I am following the same steps. when i install PCDet v0.3 by running the **_`python setup.py develop`_** . And i'm facing the above issue.

Can you tell what is the solution.

MartinHahner88, 2020-08-11:
you need to install spconv first: https://github.com/traveller59/spconv/issues/152

sshaoshuai, 2020-08-14:
Feel free to re-open it if you still have problems.

closed 2020-08-14 by sshaoshuai

natsume7, 2021-05-06:
i get the same problem but after Run cd ./dist, use pip to install generated whl file ,it work