open-mmlab/OpenPCDet #232: python3 demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --ckpt pv_rcnn_8369.pth --data_path /home/lel/Documents/DATA/kitti_velodyne/training/velodyne/000001.bin

higolel, 2020-08-10:
python3 demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --ckpt pv_rcnn_8369.pth --data_path /home/lel/Documents/DATA/kitti_velodyne/training/velodyne/000001.bin
please help me

higolel, 2020-08-10:
failed

closed 2020-08-10 by higolel

zzqjh, 2020-08-11:
could you tell us some details ? i have the same error about the demo.

zzqjh, 2020-08-11:
this is my error ,can you tell if you have sovled the error please?

Traceback (most recent call last):
  File "demo.py", line 103, in <module>
    main()
  File "demo.py", line 83, in main
    model.load_params_from_file(filename=args.ckpt, logger=logger, to_cpu=True)
  File "/home/zzqjh/3D-OT/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 322, in load_params_from_file
    raise FileNotFoundError
FileNotFoundError

RolandoAvides, 2021-02-23:
Hi!

I had the same error. Any suggestion?

Thanks!

KangChou, 2021-06-24:
`python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pt
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
/usr/local/lib/python3.6/dist-packages/traits/etsconfig/etsconfig.py:415: UserWarning: Environment variable "HOME" not set, setting home directory to /tmp
  % (environment_variable, parent_directory)
Warning: kornia is not installed correctly, please ignore this warning if you do not use CaDDN. Otherwise, it is recommended to use torch version greater than 1.2 to use kornia properly.
2021-06-25 00:02:38,810   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2021-06-25 00:02:38,924   INFO  Total number of samples: 	1
Traceback (most recent call last):
  File "demo.py", line 103, in <module>
    main()
  File "demo.py", line 83, in main
    model.load_params_from_file(filename=args.ckpt, logger=logger, to_cpu=True)
  File "/data/zk0624/pcdet_all/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 330, in load_params_from_file
    raise FileNotFoundError
FileNotFoundError
`

QAQAQAQAQAQ, 2021-11-27:
I have a similar issue. I solved it by using this 

python3 demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml \
    --ckpt ../checkpoints/pv_rcnn_8369.pth \
    --data_path ../data/demo_data

mJieg, 2022-03-29:
> 

Did you really have solved?

mJieg, 2022-03-29:
> I have a similar issue. I solved it by using this
> 
> python3 demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --ckpt ../checkpoints/pv_rcnn_8369.pth --data_path ../data/demo_data

Did you really have solved?

sharifinjf, 2022-05-24:
Hi!

I had the same error. Any suggestion?

Thanks!

thillRobot, 2024-05-09:
I had the same error. We missed step 1 in the instructions.

You need to download a pretrained model to get the `pv_rcnn_8369.pth` file first. The link is in the main README.