open-mmlab/OpenPCDet #239: muti-training error(CBGS,Nuscence)

LittleYuanzi, 2020-08-11:
Hello, when I used multi-gpus to train CBGS (nuscenes), I encountered the mistake of multi-gpus training. What is the problem? 
I can train with single-gpu.
`epochs:   0%|          | 0/20 [00:00<?, ?it/s]
epochs:   0%|          | 0/20 [00:00<?, ?it/s]Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/spawn.py", line 105, in spawn_main
    exitcode = _main(fd)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/spawn.py", line 115, in _main
    self = reduction.pickle.load(from_parent)
MemoryError
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/spawn.py", line 105, in spawn_main
    exitcode = _main(fd)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/spawn.py", line 115, in _main
    self = reduction.pickle.load(from_parent)
MemoryError
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/spawn.py", line 105, in spawn_main
    exitcode = _main(fd)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/multiprocessing/spawn.py", line 115, in _main
    self = reduction.pickle.load(from_parent)
MemoryError

           `

LittleYuanzi, 2020-08-11:
And I trained with the fellow command:
CUDA_VISIBLE_DEVICES=1,2 python -m torch.distributed.launch --nproc_per_node=2 train.py --launcher pytorch --cfg_file ./cfgs/nuscenes_models/cbgs_second_multihead.yaml --batch_size 12 --extra_tag cbgs_inital

closed 2020-08-11 by LittleYuanzi

mrqrs, 2021-05-30:
麻烦问下你是如何解决这个问题的？我也遇到了同样的问题，不知道怎么解决。