open-mmlab/OpenPCDet #240: error: command 'x86_64-linux-gnu-g++' failed with exit status 1

SupriyaB1, 2020-08-11:
ros@rosmachine:~/Downloads/OpenPCDet$ sudo python3 setup.py develop
running develop
Checking .pth file support in /usr/local/lib/python3.7/dist-packages/
/usr/bin/python3 -E -c pass
TEST PASSED: /usr/local/lib/python3.7/dist-packages/ appears to support .pth files
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
1.9.0
x86_64-linux-gnu-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_cpu.o /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms.o /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -L/home/ros/.local/lib/python3.7/site-packages/torch/lib -L/usr/local/cuda/lib64 -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda -o build/lib.linux-x86_64-3.7/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-37m-x86_64-linux-gnu.so
x86_64-linux-gnu-g++: error: /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_cpu.o: No such file or directory
x86_64-linux-gnu-g++: error: /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o: No such file or directory
x86_64-linux-gnu-g++: error: /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms.o: No such file or directory
x86_64-linux-gnu-g++: error: /home/ros/Downloads/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o: No such file or directory
error: command 'x86_64-linux-gnu-g++' failed with exit status 1
 
Hi,I am facing this issue,can anyone please help me to solve this erroe

sshaoshuai, 2020-08-14:
What's the version of PyTorch/Cuda/GCC? 
Please remove the `OpenPCDet/build` directory and try to re-install it and post the full log information to here.

sshaoshuai, 2020-08-25:
Close it since no response. Feel free to re-open it if problems still exist.

closed 2020-08-25 by sshaoshuai

yinggo, 2021-08-31:
> What's the version of PyTorch/Cuda/GCC?
> Please remove the `OpenPCDet/build` directory and try to re-install it and post the full log information to here.

I meet the same error.
Pytorch:1.7.1 CUDA11.0 GCC 5.4
I re-pulled and installed it, but get the same error, plz help.
`(pcdet) zyy@zyy-System-Product-Name:~/anaconda3/envs/pcdet/OpenPCDet$ python setup.py develop
running develop
running egg_info
creating pcdet.egg-info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
writing manifest file 'pcdet.egg-info/SOURCES.txt'
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
creating /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build
creating /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7
creating /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet
creating /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops
creating /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms
creating /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src
Emitting ninja build file /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
1.10.2
creating build/lib.linux-x86_64-3.7
creating build/lib.linux-x86_64-3.7/pcdet
creating build/lib.linux-x86_64-3.7/pcdet/ops
creating build/lib.linux-x86_64-3.7/pcdet/ops/iou3d_nms
g++ -pthread -shared -B /home/zyy/anaconda3/envs/pcdet/compiler_compat -L/home/zyy/anaconda3/envs/pcdet/lib -Wl,-rpath=/home/zyy/anaconda3/envs/pcdet/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_cpu.o /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms.o /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -L/home/zyy/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/lib -L:/usr/local/cuda-11.0/lib64 -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda -o build/lib.linux-x86_64-3.7/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-37m-x86_64-linux-gnu.so
g++: error: /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_cpu.o: No such file or directory
g++: error: /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_api.o: No such file or directory
g++: error: /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms.o: No such file or directory
g++: error: /home/zyy/anaconda3/envs/pcdet/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o: No such file or directory
error: command 'g++' failed with exit status 1
`

RendongPi, 2022-01-12:
hello, have you solved that problem?

haozy008, 2023-01-12:
I get the same error