open-mmlab/OpenPCDet #242: Confusion about model parameters

LCJHust, 2020-08-12:
Hi：
    When I trained PointPillars on KITTI dataset with default configure (80 epcoh),  I got model parameters which is about 56M, but the model zoom released (PointPillars) is 19M ,  so is the default configure the same with the one released model? Or where should I modify to get the same result with model zoom? Thank you!

sshaoshuai, 2020-08-12:
The saved model (56M) also has optimizer params, so its size is larger than the released model. But their model parameters are exactly the same.

LCJHust, 2020-08-13:
> The saved model (56M) also has optimizer params, so its size is larger than the released model. But their model parameters are exactly the same.

OK， thank you for your reply!

LCJHust, 2020-08-13:
> The saved model (56M) also has optimizer params, so its size is larger than the released model. But their model parameters are exactly the same.

Hi, I met anothor problem when I test with pretrained NuScenes 3D Object Detection Baselines. I used the default configure in [https://github.com/open-mmlab/OpenPCDet/tree/master/tools/cfgs/nuscenes_models](url), and weights updated 3 days ago. And I modified the [https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/models/backbones_2d/base_bev_backbone.py#L49](url) as your reply in anothor issue, but I still got bad results.  pp_multihead mAP：0.2967 
[image]
[pp_multihead.txt](https://github.com/open-mmlab/OpenPCDet/files/5067307/pp_multihead.txt) Can you provide some help?

sshaoshuai, 2020-08-13:
@LCJHust Yo DO NOT need to modify `base_bev_backbone.py` if you already used my updated NuScenes pretrained models.

LCJHust, 2020-08-13:
> @LCJHust Yo DO NOT need to modify `base_bev_backbone.py` if you already used my updated NuScenes pretrained models.

Thank you! I got similar results with model zooms!

closed 2020-08-14 by sshaoshuai