open-mmlab/OpenPCDet #243: Segmentation fault caused in preprocessing pipeline (py36)

tjtanaa, 2020-08-12:
I would like to setup this issue as a contribution to the community as to a solution to the following problems.

I have ran the `demo.py` under the following environment setup

Setup Environment:
Ubuntu 18.04
GCC: 7.5.0
G++: 7.5.0
GPU: RTX2060
CUDA: 10.1
CUDNN: 7.6.x
Pytorch: 1.5.0
spconv: 1.2.0 (commit abf0acf)
numba: 0.51.0rc1                pypi_0    pypi
numpy: 1.19.1           py36hbc911f0_0  
numpy-base: 1.19.1           py36hfa32c7d_0  
pcdet: 0.3.0+42a88d7            (commit 9e460da)

> python demo.py --cfg_file cfgs/kitti_models/second.yaml --ckpt second_7862.pth --data_path /home/tjtanaa/Documents/Github/ouster_example_1.14-beta.11/ouster_viz/build/2020-08-12=03_31_37_515_dir/2020-08-12=03_31_37_731.bin
2020-08-12 17:40:12,390   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2020-08-12 17:40:13,871   INFO  ==> Loading parameters from checkpoint second_7862.pth to GPU
2020-08-12 17:40:14,011   INFO  ==> Done (loaded 163/163)
Segmentation fault (core dumped)

The segmentation fault is caused the line below in `./pcdet/utils/common_utils.py` function `mask_points_by_range` ,
`mask = (points[:, 0] >= limit_range[0]) & (points[:, 0] <= limit_range[3]) & (points[:, 1] >= limit_range[1]) & (points[:, 1] <= limit_range[4])`

I fixed the issue by replacing that line with the following, 
```a1 = points[:, 0] >= limit_range[0]
a2 = (points[:, 0] <= limit_range[3])
a3 = (points[:, 1] >= limit_range[1])
a4 = (points[:, 1] <= limit_range[4])
mask = np.logical_and.reduce([a1,a2,a3,a4])

sshaoshuai, 2020-08-14:
Thank you for the contribution! 
I am not clear what's the different between the original codes and your writing?

tjtanaa, 2020-08-14:
I understand and agree to your question. 

Given that you have tested in python 3.6 environment, I suppose it is not a bug arise from different version of python. One possible answer to this is that it might be due to the error in the conda environment that I have created. I have once downgrade the conda environment from python3.7 to 3.6 (without deleting that environment and create a new environment.). This might have led to the packages not being updated (py37 -> py36) thus causing the overloaded '&' to cause 'segmentation fault (core dump)'.

Currently, I avoid making changes to the code and create a new conda environment (python 3.7 instead) and pytorch 1.5.1, spconv 1.2.1. It works now.

sshaoshuai, 2020-08-17:
Ok, thank you for the clarification. I agree with you and it is unresonable and may depend on the environment. 
I am closing it now and feel free to re-open it if there still exists the problem.

closed 2020-08-17 by sshaoshuai