open-mmlab/OpenPCDet #244: Can not find file kitti_dbinfos_train.pkl

davidwang200099, 2020-08-13:
I tried to run 'python train.py --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml'. But it failed. The error log is:
'''
2020-08-13 15:09:49,270   INFO  **********************Start logging**********************
2020-08-13 15:09:49,270   INFO  CUDA_VISIBLE_DEVICES=0
2020-08-13 15:09:49,270   INFO  cfg_file         ./cfgs/kitti_models/pv_rcnn.yaml
2020-08-13 15:09:49,270   INFO  batch_size       2
2020-08-13 15:09:49,270   INFO  epochs           80
2020-08-13 15:09:49,270   INFO  workers          8
2020-08-13 15:09:49,270   INFO  extra_tag        default
2020-08-13 15:09:49,270   INFO  ckpt             None
2020-08-13 15:09:49,271   INFO  pretrained_model None
2020-08-13 15:09:49,271   INFO  launcher         none
2020-08-13 15:09:49,271   INFO  tcp_port         18888
2020-08-13 15:09:49,271   INFO  sync_bn          False
2020-08-13 15:09:49,271   INFO  fix_random_seed  False
2020-08-13 15:09:49,271   INFO  ckpt_save_interval 1
2020-08-13 15:09:49,271   INFO  local_rank       0
2020-08-13 15:09:49,271   INFO  max_ckpt_save_num 30
2020-08-13 15:09:49,271   INFO  merge_all_iters_to_one_epoch False
2020-08-13 15:09:49,271   INFO  set_cfgs         None
2020-08-13 15:09:49,271   INFO  max_waiting_mins 0
2020-08-13 15:09:49,271   INFO  start_epoch      0
2020-08-13 15:09:49,271   INFO  save_to_file     False
2020-08-13 15:09:49,271   INFO  cfg.ROOT_DIR: /home/zehao/PV-RCNN/OpenPCDet
2020-08-13 15:09:49,271   INFO  cfg.LOCAL_RANK: 0
2020-08-13 15:09:49,271   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-08-13 15:09:49,271   INFO
cfg.DATA_CONFIG = edict()
2020-08-13 15:09:49,271   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-08-13 15:09:49,271   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2020-08-13 15:09:49,272   INFO
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-08-13 15:09:49,272   INFO
cfg.DATA_CONFIG.INFO_PATH = edict()
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-08-13 15:09:49,272   INFO
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-08-13 15:09:49,272   INFO
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2020-08-13 15:09:49,272   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2020-08-13 15:09:49,272   INFO
cfg.MODEL = edict()
2020-08-13 15:09:49,273   INFO  cfg.MODEL.NAME: PVRCNN
2020-08-13 15:09:49,273   INFO
cfg.MODEL.VFE = edict()
2020-08-13 15:09:49,273   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2020-08-13 15:09:49,273   INFO
cfg.MODEL.BACKBONE_3D = edict()
2020-08-13 15:09:49,273   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2020-08-13 15:09:49,273   INFO
cfg.MODEL.MAP_TO_BEV = edict()
2020-08-13 15:09:49,273   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2020-08-13 15:09:49,273   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2020-08-13 15:09:49,273   INFO
cfg.MODEL.BACKBONE_2D = edict()
2020-08-13 15:09:49,273   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2020-08-13 15:09:49,273   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2020-08-13 15:09:49,273   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
2020-08-13 15:09:49,273   INFO  cfg.MODEL.BACKBONE_2D.NUM_FILTERS: [128, 256]
[... 162 lines cut ...]
2020-08-13 15:09:49,280   INFO  cfg.OPTIMIZATION.LR_WARMUP: False
2020-08-13 15:09:49,281   INFO  cfg.OPTIMIZATION.WARMUP_EPOCH: 1
2020-08-13 15:09:49,281   INFO  cfg.OPTIMIZATION.GRAD_NORM_CLIP: 10
2020-08-13 15:09:49,281   INFO  cfg.TAG: pv_rcnn
2020-08-13 15:09:49,281   INFO  cfg.EXP_GROUP_PATH: cfgs/kitti_models
Traceback (most recent call last):
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 112, in main
    total_epochs=args.epochs
  File "/home/zehao/PV-RCNN/OpenPCDet/pcdet/datasets/__init__.py", line 49, in build_dataloader
    logger=logger,
  File "/home/zehao/PV-RCNN/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 23, in __init__
    dataset_cfg=dataset_cfg, class_names=class_names, training=training, root_path=root_path, logger=logger
  File "/home/zehao/PV-RCNN/OpenPCDet/pcdet/datasets/dataset.py", line 32, in __init__
    ) if self.training else None
  File "/home/zehao/PV-RCNN/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 23, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "/home/zehao/PV-RCNN/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 31, in gt_sampling
    logger=self.logger
  File "/home/zehao/PV-RCNN/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 21, in __init__
    with open(str(db_info_path), 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/zehao/PV-RCNN/OpenPCDet/data/kitti/kitti_dbinfos_train.pkl'
'''
I have noticed [this issue](https://github.com/open-mmlab/OpenPCDet/issues/104) and modified the file, but there is still such error.

closed 2020-08-13 by davidwang200099

hcyoo93, 2020-08-19:
@davidwang200099 
I am facing the same issue. Did you resolve this issue?

davidwang200099, 2020-08-19:
Did you run ```python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml``` as [GETTING_STARTED.md](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md) requires you to do?

hcyoo93, 2020-08-19:
@davidwang200099 
Thanks for the reply!
I did it, but I did not recognize that error in execution of "creat_kitti_infos" results in skipping create_groundtruth_database.

Since "create_groundtruth_database" generates 'kitti_dbinfos_train.pkl', I encountered the error!
It is resolved now.

wind19970924, 2020-10-05:
> @davidwang200099
> Thanks for the reply!
> I did it, but I did not recognize that error in execution of "creat_kitti_infos" results in skipping create_groundtruth_database.
> 
> Since "create_groundtruth_database" generates 'kitti_dbinfos_train.pkl', I encountered the error!
> It is resolved now.
how to slove the error in execution of "creat_kitti_infos" results in skipping create_groundtruth_database,thanks a lot