open-mmlab/OpenPCDet #245: Error: roiaware_pool3d_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: __cudaRegisterFatBinaryEnd

Feng-HC, 2020-08-13:
Hello
I meet two problems.
First， when I run "python test.py", the error is raised:

-------------------------------error information start----------------------------------------
File "/home/fhc/.conda/envs/pytorch1.1/lib/python3.6/site-packages/numba/cuda/decorators.py", line 96, in kernel_jit
    kernel.bind()
  File "/home/fhc/.conda/envs/pytorch1.1/lib/python3.6/site-packages/numba/cuda/compiler.py", line 489, in bind
    self._func.get()
  File "/home/fhc/.conda/envs/pytorch1.1/lib/python3.6/site-packages/numba/cuda/compiler.py", line 370, in get
    ptx = self.ptx.get()
  File "/home/fhc/.conda/envs/pytorch1.1/lib/python3.6/site-packages/numba/cuda/compiler.py", line 342, in get
    **self._extra_options)
  File "/home/fhc/.conda/envs/pytorch1.1/lib/python3.6/site-packages/numba/cuda/cudadrv/nvvm.py", line 464, in llvm_to_ptx
    libdevice = LibDevice(arch=opts.get('arch', 'compute_20'))
  File "/home/fhc/.conda/envs/pytorch1.1/lib/python3.6/site-packages/numba/cuda/cudadrv/nvvm.py", line 340, in __init__
    raise RuntimeError(MISSING_LIBDEVICE_FILE_MSG.format(arch=arch))
RuntimeError: Missing libdevice file for compute_50.
Please ensure you have package cudatoolkit 7.5.
Install package by:

    conda install cudatoolkit=7.5
-------------------------------error information end-------------------------------------

Then I try to run "conda install numba" to resolve it.
But another problem occurs when I run "python test.py".

-------------------------------error information start-----------------------------------------------
Traceback (most recent call last):
  File "test.py", line 11, in <module>
    from pcdet.datasets import build_dataloader
  File "/home/fhc/OpenPCDet/pcdet/datasets/__init__.py", line 3, in <module>
    from .dataset import DatasetTemplate
  File "/home/fhc/OpenPCDet/pcdet/datasets/dataset.py", line 5, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/home/fhc/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 3, in <module>
    from . import augmentor_utils, database_sampler
  File "/home/fhc/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 3, in <module>
    from ...utils import box_utils
  File "/home/fhc/OpenPCDet/pcdet/utils/box_utils.py", line 4, in <module>
    from ..ops.roiaware_pool3d import roiaware_pool3d_utils
  File "/home/fhc/OpenPCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 5, in <module>
    from . import roiaware_pool3d_cuda
ImportError: /home/fhc/OpenPCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: __cudaRegisterFatBinaryEnd
-------------------------------error information end-------------------------------------

my version information is :
pytorch 1.1
cuda 10.1
cudatoolkit 10.0
pcdet v2.0

I try to reinstall pytorch=1.1 and cudatoolkit=9.0, remove the "OpenPCDet/build" and rebuild openpcdet.
But the same error also occurs.

Note: Before the first problem occurs, the second problem does not occur.

Feng-HC, 2020-08-13:
Error: roiaware_pool3d_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: __cudaRegisterFatBinaryEnd

sshaoshuai, 2020-08-14:
Please refer to here https://github.com/open-mmlab/mmdetection/issues/385. 
You may need to  "check the compatibility of your Pytorch and Cuda version".

closed 2020-08-17 by sshaoshuai