open-mmlab/OpenPCDet #247: Runtime error: subprocess.CalledProcessError

davidwang200099, 2020-08-14:
I tried to run ```./scripts/dist_train.sh 2 --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml```. The training stage finished smoothly. Then comes the evaluation stage. The following information was given:
```
2020-08-14 20:10:03,581   INFO  *************** Performance of EPOCH 79 *****************
2020-08-14 20:10:03,581   INFO  Generate label finished(sec_per_example: 0.0442 second).
2020-08-14 20:10:03,581   INFO  recall_roi_0.3: 0.963554
2020-08-14 20:10:03,581   INFO  recall_rcnn_0.3: 0.964009
2020-08-14 20:10:03,582   INFO  recall_roi_0.5: 0.917198
2020-08-14 20:10:03,582   INFO  recall_rcnn_0.5: 0.923633
2020-08-14 20:10:03,582   INFO  recall_roi_0.7: 0.689123
2020-08-14 20:10:03,582   INFO  recall_rcnn_0.7: 0.731264
2020-08-14 20:10:03,584   INFO  Average predicted number of objects(3769 samples): 10.271
```
Then there is some runtime error, and the log is:
```
Traceback (most recent call last):
  File "/home/zehao/anaconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/zehao/anaconda3/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/zehao/anaconda3/lib/python3.7/site-packages/torch/distributed/launch.py", line 261, in <module>
    main()
  File "/home/zehao/anaconda3/lib/python3.7/site-packages/torch/distributed/launch.py", line 257, in main
    cmd=cmd)
subprocess.CalledProcessError: Command '['/home/zehao/anaconda3/bin/python', '-u', 'train.py', '--local_rank=1', '--launcher', 'pytorch', '--cfg_file', './cfgs/kitti_models/pv_rcnn.yaml']' died with <Signals.SIGSEGV: 11>.
```
I added some lines of code to output some logs, and located the error in function``` evaluation``` in file ```pcdet/datasets/kitti/kitti_dataset.py```. I added the code like this:
```
def evaluation(self, det_annos, class_names, **kwargs):
        print("Special case:")
        if 'annos' not in self.kitti_infos[0].keys():
            return None, {}
        print("Special case ends.")

        from .kitti_object_eval_python import eval as kitti_eval

        print("Deepcopying")
        eval_det_annos = copy.deepcopy(det_annos)
        eval_gt_annos = [copy.deepcopy(info['annos']) for info in self.kitti_infos]

        print("Deepcopy finished.")
        ap_result_str, ap_dict = kitti_eval.get_official_eval_result(eval_gt_annos, eval_det_annos, class_names)

        print("get_official_eval_result")
        return ap_result_str, ap_dict
```
The error log is:
```
Special case:
Special case ends.
```
Then probably it can be included that the error happens when trying to deepcopy. But what may cause this error and how to fix it?

sshaoshuai, 2020-08-14:
I think it may happen at the import line. Maybe you could try to import it separately for better bug localization.

davidwang200099, 2020-08-15:
I moved that ''import...''' code after the '''deepcopy''' code and run evaluation again, and find that the segmentation fault happens when trying to import.

closed 2020-08-16 by davidwang200099