open-mmlab/OpenPCDet #252: Evaluate own Dataset

josy43, 2020-08-17:
Hello,
this repo works very good for me and I already build my own dataset in Kitti format and the labels. So where do I put my labels so I can see the score for test.py. I need to see the score between my labels and the predicted labels.

kamillasak, 2020-08-17:
@josy43 how you make labels and that files in Kitti format?

jihanyang, 2020-08-17:
Have you created infos for you own dataset?

josy43, 2020-08-17:
> Have you created infos for you own dataset?

Yes that was the point.
So now I try to get a score for my evaluation dataset, but I just got zeros everywhere. 
My Label looks like this:
Car 0 0 0 0 0 0 0 1.7759236998670875 1.8197457766280374 4.122105387415043 6.6118077507298025 2.1935728923376425 20.66966086726571 1.5385849784507857

and the predicted label like this:
Car -1 -1 -5.0051 801.8121 169.6522 904.2530 228.0073 1.4932 1.6120 4.0451 6.7107 1.4110 20.5002 -4.69365

I dont have any information about the 7 values after the type value. I just need to know the info about the bounding box. 
So how can I get th escore for my data.

Info about Kitti File Format :  https://github.com/bostondiditeam/kitti/blob/master/resources/devkit_object/readme.txt

jihanyang, 2020-08-18:
#198

closed 2020-08-25 by sshaoshuai