open-mmlab/OpenPCDet #254: undefined symbol: THPVariableClass

stratomaster31, 2020-08-18:
I've installed OpenPCDet following **install.md** on the **spconv docker image**. Under python3.8 everything goes well, but under python3.6 I get:

    ImportError: /home/ML_tools/python/pcdet_api/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-36m-x86_64-linux- 
    gnu.so: undefined symbol: THPVariableClass

Both python3.8 and python3.6 I have the same version of pytorch;

          print(torch.__version__)
          1.6.0+cu101

          print(torchvision.__version__)
          0.7.0+cu101

Thank you very much!

closed 2020-08-18 by stratomaster31

twn29004, 2021-06-24:
Hello, Can you tell me how you solve it?

yinggo, 2021-07-03:
I meet the same problem, could anyone can help?