open-mmlab/OpenPCDet #256: Other Classes

josy43, 2020-08-18:
According to the website of Kitti Benchmark there are more classes then Car, Pedestrian and Cyclist, for example Tram and Truck. Is there any pretrained model which considers also other classes ?

sshaoshuai, 2020-08-18:
No, you could modify the config file and train them by yourself.

josy43, 2020-08-18:
Thanks for the fast reply. Do I just have to add the classes in CLASS_NAMES in the yaml file? 
What does Anchor_Generator_CONFIG mean ?
And a question about the visualization: How do I display the classname next to the confidence value ?

Thanks

sshaoshuai, 2020-08-20:
You also need to modify other parts that have the class-related settings, such as the GT_sampling augmentation, anchor setting. 
The Anchor_Generator_CONFIG aims to set anchors for each class.  

I suggest that you could refer to the nuscenes configs to see how to detect more classes by comparing the configs between KITTI and NuScenes. 

For visualization, you need to explore the mayavi tools and add the features by yourself.

closed 2020-08-25 by sshaoshuai

daeunni, 2022-01-10:
> According to the website of Kitti Benchmark there are more classes then Car, Pedestrian and Cyclist, for example Tram and Truck. Is there any pretrained model which considers also other classes ?

@josy43  
Have you solved this problem? I'm thinking about the same problem.

How did you se**t Anchor_Generator_CONFIG** for classes **other** than Car, Ped, and Cyclist?