open-mmlab/OpenPCDet #258: KITTI second_multihead.yaml

sasupe, 2020-08-20:
Hi Team,
Thank you providing this repository!

Under tools/cfgs/kitti_models/, second_multihead.yaml is provided but no pre-trained model is given. Also I guess training a multi-head with KITTI is futile as there are no labels available.

Why is the multihead config provided under KITTI? Can we use the nuscenes trained model with this config file? 

I was able to run the nuscenes trained model on KITTI using the cbgs_second_multihead.yaml and using kitti_dataset.yaml. I changed the POINT_CLOUD_RANGE and VOXEL_SIZE to what nuscenes_dataset.yaml had. The results look reasonable.

Is there a correct way to run second_multihead on KITTI?

Thank You!

sshaoshuai, 2020-08-25:
@sasupe , Yes, it is correct. But you should notice that your KITTI input is correct, since the provided NuScenes models are based on 10frames for detection. Also I think the performance should not be good enough when directly adopting it on KITTI dataset since there is a domain gap between two datasets.

closed 2020-09-03 by sshaoshuai

kadebudongtan, 2022-03-02:
> Hi Team, Thank you for providing this repository!
> 
> I am trying to run second-multihead trained with nuScenes on KITTI to get 360 degree detections.
> 
> I was able to run the nuScenes trained model on KITTI using the cbgs_second_multihead.yaml and using kitti_dataset.yaml. I changed the POINT_CLOUD_RANGE and VOXEL_SIZE to what nuscenes_dataset.yaml had. The results look reasonable.
> 
> Is there a correct way to run a nuScenes trained second_multihead on KITTI to get 360 degree detections?
> 
> Thank You!

Hey, you said your result is reasonable. could you pls share your config files? I just use the provided cbgs_pp_multihead model to visualize the kitti data, the results are changeable and terrible.