open-mmlab/OpenPCDet #262: Training on other datasets - requirements

jacoblambert, 2020-08-24:
With neither kitti_dataset.py or nuscenes_dataset.py being documented or easily readable, it would be really nice to know what the requirements of:
- dbinfos.pkl
- infos_trains.pkl
- infos_val.pkl
- gt_database/
or some kind of minimal custom_dataset.py template we could follow towards training on a custom dataset.

The easiest way I'm seeing right now is to convert any dataset to KITTI format and use the KITTI loader but it converting datasets into an old, less than ideal format. I also wonder if it is sufficient to adjust POINT_CLOUD_RANGE and set FOV_POINTS_ONLY to False use full point cloud annotations.

Thank you in advance for your assistance with this.

jihanyang, 2020-08-24:
Hello,
As for us, we will re-implement methods in kitti_datasets. Actually, you only think about 3 parts: 1. `__getitem__` 2. create infos 3. evaluation

You can adjust POINT_CLOUD_RANGE and set FOV_POINTS_ONLY to False to use full point cloud, but please make sure use suitable voxel size.

jacoblambert, 2020-08-25:
Thanks for your reply @jihanyang .

I'm wondering about correct frame of reference usage. I'm guessing the "unified coordinate frame" is basically the LiDAR frame? So I should ensure pointcloud and 3D Bbox are in the same frame.

In KITTI dataset, the labels are originally in 3D camera frame I believe so there must be some tf at some point in the kitti_dataset.py code, not sure where as there is a lot going on. Did you remove this part of the code (if so could you point me where) or did you create fake /calib files creating a unity transform between camera and lidar frame?

Thanks again for your help

chasingw, 2020-09-04:
@jacoblambert hi just remove code about calib, because it transform to LiDAR frame in final `gt_boxes` used in dataset.py, for me, it worked for train

closed 2020-10-19 by sshaoshuai

bingo830422, 2020-11-03:
@chasingw, hello, could you please share the calib-removed dataset.py with me. I've encounter error about dependent variables while modifying it, and not know where went wrong.

eriche2016, 2020-11-03:
Hi, @chasingw  how do you evaluate your model?

Petros626, 2024-10-01:
Still no response here?