open-mmlab/OpenPCDet #265: Question for projection of points to rectified image

DrBaer, 2020-08-25:
https://github.com/open-mmlab/OpenPCDet/blob/9e460da23163dffcca007ac8725221257cf36d55/pcdet/utils/calibration_kitti.py#L82

Why is pts_rect_hom used here and not pts_2d_hom? The values are not equal due to a non-zero entry in self.P2.T[3, 2]

sshaoshuai, 2020-08-27:
Thank you for the bug report. 

It seems you are right, and it should be pts_2d_hom[:, 2], but it has minor effects on the 3D mAP.  
Have you tried to fix it? What's the performance difference after using the updated one?

DrBaer, 2020-10-01:
I guess it does not have a big impact as it affected only a minority of points. I evaluate it if I find the time. I guess the impact of model initialization is bigger than this change as there seems to be 1-2% mAP change with multiple trainings.

closed 2022-03-08 by github-actions[bot]