open-mmlab/OpenPCDet #266: perform inference over full (360deg) view

hareld10, 2020-08-25:
Hi,

Is it possible?

Couldn't find any mention over the configurations.

Best,

sshaoshuai, 2020-08-25:
Please refer to the NuScenes configs, since the NuScenes configs already detect over the full 360 degree view.

tlind, 2020-08-28:
Does this require retraining, or is it possible to train on limited FOV and run inference over the full FOV? Thanks!

jacoblambert, 2020-09-03:
If you want to perform KITTI model on full PCD, I don't think the performance will be good changing the PCD range/voxel size without retraining. My approach was to run inference on original pointcloud -> flip the pcd 180 degrees -> flip the obtained labels back to the original reference and concatenate with the original labels. Sometimes you can get duplicate labels at the edges so you need to run NMS again. It works well, in fact so far my experience is KITTI models work much better on real data than NuScenes models (because of sweeps and velocities, I imagine). He's the results on a KITTI test sample, full 360 degrees:

[image]

closed 2020-10-19 by sshaoshuai

ryontang, 2021-02-13:
> If you want to perform KITTI model on full PCD, I don't think the performance will be good changing the PCD range/voxel size without retraining. My approach was to run inference on original [...]

Hi,@jacoblambert, thanks for your share, I am a newer in using this rep, could you plz share some code of your work that can perform KITTI model on full PCD.

Tom698, 2022-09-07:
> If you want to perform KITTI model on full PCD, I don't think the performance will be good changing the PCD range/voxel size without retraining. My approach was to run inference on original [...]

Good job Ryontang! I want to konw how to  flip the pcd 180 degrees. Would you give me some advice?

Petros626, 2025-03-31:
@jacoblambert can you share the code to do the same?

jacoblambert, 2025-05-20:
@ryontang @Petros626 Sorry for missing your prior comments. It's been quite a while so I don't have the original code, but you can modify demo.py to run twice, once on the original PCD and once on the flipped pcd (e.g. x -> -x) and then apply the opposite transformation to the resulting bounding box.

To answer your questions @Petros626 

> 1. Did you flipped the original limited (only front view from lidar origin) pc around the x-axis (lidar frame), bc the image you posted doesn't seem like the mirrored scene.

I flip the resulting bounding box back in the original, full 360 degree scene.

> 2. How to obtain gt-labels for a full 360° scene for training. For KITTI only labels in the FoV of the camera are available.
There's no way to get full labels, this is only for inference. Maybe [KITTI 360](https://www.cvlibs.net/datasets/kitti-360/documentation.php)?

> 3. Do you got some repo or code where your implement the extended inference?
Nope sorry, I'm not sure where I implemented this anymore.

Petros626, 2025-05-20:
@jacoblambert thanks for the answer, I still struggle with the imagination of a flipped point cloud, because it would be the same as the front view just mirrored with bounding boxes.. 

I thought about making the whole 360° of the LiDAR data available (like here https://github.com/open-mmlab/OpenPCDet/issues/266#issuecomment-680142134) to the detector, so that it can train and do inference on it. Predicted bounding boxes would be then for the front view (everything before sensor origin) and rear view (behind sensor origin).