open-mmlab/OpenPCDet #267: about Data Augmentation

davidwang200099, 2020-08-26:
I noticed that several functions, such as ```random_world_flip```, ```random_world_rotation```, and ```random_world_scaling``` were implemented in ```data_augmentor.py```. However, I read through the code, only to find no function invoking them to carry out data augmentation. I put some lines of code in the three functions above to get some log, and found that data augmentation is carried out when trying to train each epoch. Can some functions in pytorch invoke the data augmenter to carry out data augmentation?

jihanyang, 2020-08-26:
Please check `prepare_data` in `dataset.py`.

davidwang200099, 2020-08-27:
Thanks! I find it!

closed 2020-08-27 by davidwang200099

Petros626, 2024-04-18:
@davidwang200099 could you tell me how I can do data augmentation on a dataset e.g. like KITTI?