open-mmlab/OpenPCDet #268: Error When Run QuickDemo

DSwutang, 2020-08-26:
When I run QuickDemo, I got this error report. 

> 2020-08-26 19:15:11,525    INFO    -------------Quick Demo of OpenPCDet---------------
2020-08-26 19:15:11,649     INFO    Total number of samples:             1
2020-08-26 19:15:13,296     INFO     ==> Loading parameters from checkpoint pv_rcnn_8369.pth to CPU
2020-08-26 19:15:13,946     INFO     ==> Done  (loaded 367/367)
2020-08-26 19:15:13,982     INFO     Visualized sample index:              1
X  Error of failed request:    BadWindow  (invalid Window parameter)
    Major opcode of failed request:    40 (X_TranslateCoords)
    Resource id in failed request:    0x0
    Serial number of failed request:    9
    Current serial number in output stream:    9

I type it= =.
Ubuntu16.04 / pytorch1.1 / spconv1.0 / cuda11.0 / cudnn8.0
That's my first time using NVIDIA to run code. I wonder if my NVIDIA settings have some problems.
Thanks a lot.

sshaoshuai, 2020-09-03:
Maybe you need to run this visualization codes in your local machine with a screen.

MartinHahner88, 2020-09-04:
You can maybe have a look at how I solved it.
https://github.com/MartinHahner88/OpenPCDet/blob/master/tools/demo.py#L19-L24

I run the inference on a headless server and then visualize the `.pkl` files locally where a monitor is attached with the following file
https://github.com/MartinHahner88/OpenPCDet/blob/master/tools/visual_utils/visualize.py

Hope this helps.

YangGou, 2020-10-26:
> When I run QuickDemo, I got this error report.
>
>
> I type it= =.
> Ubuntu16.04 / pytorch1.1 / spconv1.0 / cuda11.0 / cudnn8.0
> That's my first time using NVIDIA to run code. I wonder if my NVIDIA [...]

I meet the same problem,have you solved it?

YangGou, 2020-10-26:
> You can maybe have a look at how I solved it.
> https://github.com/MartinHahner88/OpenPCDet/blob/master/tools/demo.py#L19-L24
>
> I run the inference on a headless server and then visualize the `.pkl` [...]

the link is invalid,can you upload it again,thanks.

MartinHahner, 2020-10-26:
contents of demo.py

```python
import os
import glob
import torch
import pickle
import shutil
import argparse

import numpy as np

from tqdm import tqdm
from pathlib import Path

from pcdet.config import cfg, cfg_from_yaml_file
from pcdet.datasets import DatasetTemplate
from pcdet.models import build_network, load_data_to_gpu
from pcdet.utils import common_utils

if os.name == 'posix' and "DISPLAY" not in os.environ:
    headless_server = True
else:
    headless_server = False
    from tools.visual_utils import visualize_utils as V
    import mayavi.mlab as mlab

def parse_config():

    parser = argparse.ArgumentParser(description='arg parser')

    parser.add_argument('-m', '--method', type=str, default='pv_rcnn', help='specify the method')
    parser.add_argument('-d', '--dataset', type=str, default='KITTI', help='specify the dataset')
    parser.add_argument('-r', '--dataset_root', type=str, help='specify the dataset root',
                        default='/srv/beegfs-benderdata/scratch/tracezuerich/data/datasets/')
    parser.add_argument('-c', '--copy_image', type=bool, help='specify if images should be copied to DEMO folder',
                        default=True)

    args = parser.parse_args()

    cfg_file = get_cfg_file(args.method)

    cfg_from_yaml_file(cfg_file, cfg)

    return args, cfg
```
...

```python
    with torch.no_grad():

        for idx, data_dict in tqdm(enumerate(demo_dataset), total=len(demo_dataset)):

            data_dict = demo_dataset.collate_batch([data_dict])
            load_data_to_gpu(data_dict)
            pred_dicts, _ = model.forward(data_dict)

            if headless_server:

                image_name = demo_dataset.sample_file_list[idx].split('/')[-1].replace(demo_dataset.ext, '.png')

                image_source_path = image_path / image_name

                if args.copy_image and image_source_path.exists():
                    image_destination_path = export_dir / image_name
                    shutil.copyfile(image_source_path, image_destination_path)

                data_dict_cpu = {}

                for key, value in data_dict.items():
                    if key == 'points':
                        data_dict_cpu[key] = value.cpu().numpy()

                with open(f'{export_dir}/data_dict_{idx+1:06d}.pkl', 'wb') as f:
                    pickle.dump(data_dict_cpu, f, pickle.HIGHEST_PROTOCOL)

                for pred_dict in pred_dicts:

                    for key, value in pred_dict.items():
                        if isinstance(pred_dict[key], torch.Tensor):
                            pred_dict[key] = value.cpu().numpy()

                with open(f'{export_dir}/pred_dicts_{idx + 1:06d}.pkl', 'wb') as f:
                    pickle.dump(pred_dicts, f, pickle.HIGHEST_PROTOCOL)

            else:
                V.draw_scenes(
                    points=data_dict['points'][:, 1:], ref_boxes=pred_dicts[0]['pred_boxes'],
                    ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
                )
                mlab.show(stop=True)

    logger.info('Demo done.')
```

MartinHahner, 2020-10-26:
content of viusalize.py

```python
import os
import torch
import pickle
import pandas
import numpy as np
from pathlib import Path
import mayavi.mlab as mlab

box_colormap = [[1, 1, 1],
                [0, 1, 0],
                [0, 1, 1],
                [1, 1, 0]]

def check_numpy_to_torch(x):
    if isinstance(x, np.ndarray):
        return torch.from_numpy(x).float(), True
    return x, False

def rotate_points_along_z(points, angle):
    """
    Args:
        points: (B, N, 3 + C)
        angle: (B), angle along z-axis, angle increases x ==> y
    Returns:

    """
    points, is_numpy = check_numpy_to_torch(points)
    angle, _ = check_numpy_to_torch(angle)

    cosa = torch.cos(angle)
    sina = torch.sin(angle)
    zeros = angle.new_zeros(points.shape[0])
    ones = angle.new_ones(points.shape[0])
    rot_matrix = torch.stack((
        cosa,  sina, zeros,
        -sina, cosa, zeros,
        zeros, zeros, ones
    ), dim=1).view(-1, 3, 3).float()
    points_rot = torch.matmul(points[:, :, 0:3], rot_matrix)
    points_rot = torch.cat((points_rot, points[:, :, 3:]), dim=-1)
    return points_rot.numpy() if is_numpy else points_rot

def visualize_pts(pts, bgcolor=(0, 0, 0), fgcolor=(1.0, 1.0, 1.0),
                  color_feature=None, size=(600, 600), draw_origin=True, title=None):

    min_height_mask = pts[:, 2] > -1.5
    pts = pts[min_height_mask, :]

    fig = mlab.figure(figure=title, bgcolor=bgcolor, fgcolor=fgcolor, engine=None, size=size)

    if color_feature:
        mlab.points3d(pts[:, 0], pts[:, 1], pts[:, 2], -pts[:, color_feature], mode='point',
                      colormap='jet', scale_factor=1, figure=fig)
    else:
        mlab.points3d(pts[:, 0], pts[:, 1], pts[:, 2], mode='point',
                      colormap='jet', scale_factor=1, figure=fig)

    if draw_origin:
        cylinder = mlab.points3d(0, 0, 0, color=(1, 1, 1), mode='cylinder', scale_mode='vector', scale_factor=3)
        cylinder.glyph.glyph_source._trfm.transform.rotate_x(90)
        mlab.plot3d([0, 3], [0, 0], [0, 0], color=(0, 0, 1), tube_radius=0.1)
        mlab.plot3d([0, 0], [0, 3], [0, 0], color=(0, 1, 0), tube_radius=0.1)
        mlab.plot3d([0, 0], [0, 0], [0, 3], color=(1, 0, 0), tube_radius=0.1)

    return fig

def draw_multi_grid_range(fig, grid_size=20, bv_range=(-60, -60, 60, 60)):
[... 145 lines cut ...]

    df = pandas.read_csv('../../data/kitti/ImageSets/test.txt', header=None)

    index = 5896 # has to be -1 of index in all.txt

    while True:

        sample_id = f'{index+1:06d}'
        current_recording = f'{df.values[index][0]}_{df.values[index][1]}'

        title = f'# {sample_id}/{num_samples} - {current_recording}'

        with open(f'{import_dir}/data_dict_{sample_id}.pkl', 'rb') as f:
            data_dict = pickle.load(f)

        with open(f'{import_dir}/pred_dicts_{sample_id}.pkl', 'rb') as f:
            pred_dicts = pickle.load(f)

        scene = draw_scene(points=data_dict['points'][:, 1:], ref_boxes=pred_dicts[0]['pred_boxes'],
                            ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels'],
                            title=title, confidence=0.6)

        mlab.show(stop=False)
        index += 1 % len(data)
```

yuandm, 2020-11-20:
I met the same problem? how to fix?

riceleslie, 2020-11-20:
> I met the same problem? how to fix?
you could try to avoid running code related to 'mlab'. and try to show your picture in local.
you just need to save 4 tensors(points, ref_boxes, ref_scores, ref_labels) to files. then copy these files to local and get this 4 tensors from files. feed these tensor into function  'draw_scene'

closed 2022-03-08 by github-actions[bot]