open-mmlab/OpenPCDet #269: Error When Run QuickDemo

zzqjh, 2020-08-27:
When I run QuickDemo, I got this error report.
could you have some suggestion for me please？

python3 demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --ckpt pv_rcnn_8369.pth --data_path /home/zzqjh/3D-OT/OpenPCDet/data/kitti/training/velodyne/000004.bin
2020-08-27 19:46:55,152   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2020-08-27 19:46:55,250   INFO  Total number of samples: 	1
Traceback (most recent call last):
  File "demo.py", line 103, in <module>
    main()
  File "demo.py", line 83, in main
    model.load_params_from_file(filename=args.ckpt, logger=logger, to_cpu=True)
  File "/home/zzqjh/3D-OT/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 322, in load_params_from_file
    raise FileNotFoundError
FileNotFoundError

sshaoshuai, 2020-09-03:
You should download the pretrained models and use `--ckpt` to specify the path of downloaded model.

zzqjh, 2020-09-03:
yeah,thank you very much!And I have run the demo and the test.py successfully,haha

closed 2020-09-03 by zzqjh

junjiesong0427, 2024-06-23:
024-06-23 22:20:08,709   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2024-06-23 22:20:08,710   INFO  Total number of samples:        1
Traceback (most recent call last):
  File "tools/demo.py", line 112, in <module>
    main()
  File "tools/demo.py", line 90, in main
    model.load_params_from_file(filename=args.ckpt, logger=logger, to_cpu=True)
  File "/root/PointCloudDet3D/pcdet/models/detectors/detector3d_template.py", line 363, in load_params_from_file
    raise FileNotFoundError
FileNotFoundError
训练后demo还是报错可以帮解决下吗