open-mmlab/OpenPCDet #273: RuntimeError: CUDA error: invalid device function

hughlee815, 2020-09-01:
hi,
when i run test.py, it throws such an error:
Error!
Traceback (most recent call last):
  File "test.py", line 200, in <module>
    main()
  File "test.py", line 196, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 62, in eval_single_ckpt
    eval_utils.eval_one_epoch(
  File "/home/nio/Desktop/pointpillar/tools/eval_utils/eval_utils.py", line 57, in eval_one_epoch
    pred_dicts, ret_dict = model(batch_dict)
  File "/home/nio/anaconda3/lib/python3.8/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/nio/Desktop/pointpillar/pcdet/models/detectors/pointpillar.py", line 21, in forward
    pred_dicts, recall_dicts = self.post_processing(batch_dict)
  File "/home/nio/Desktop/pointpillar/pcdet/models/detectors/detector3d_template.py", line 261, in post_processing
    recall_dict = self.generate_recall_record(
  File "/home/nio/Desktop/pointpillar/pcdet/models/detectors/detector3d_template.py", line 298, in generate_recall_record
    iou3d_rcnn = iou3d_nms_utils.boxes_iou3d_gpu(box_preds[:, 0:7], cur_gt[:, 0:7])
  File "/home/nio/Desktop/pointpillar/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 71, in boxes_iou3d_gpu
    overlaps_h = torch.clamp(min_of_max - max_of_min, min=0)
RuntimeError: CUDA error: invalid device function

looking forward to your reply...

deanzhanggit, 2020-09-10:
same problem

MartinHahner88, 2020-09-11:
on which GPU are you trying to run `test.py`? Is the GPU based on the _Maxwell_ or on a newer architecture?

deanzhanggit, 2020-09-11:
I use Tesla V100 and CUDA10.1

MartinHahner88, 2020-09-11:
Are you able to try CUDA 9.2?

deanzhanggit, 2020-09-11:
Thank you for your reply, I'll try it later.

deanzhanggit, 2020-09-12:
I tried CUDA 9.2, problem solved！

closed 2020-09-15 by hughlee815

QAQAQAQAQAQ, 2021-11-27:
Hi,

When I run demo.py I got a similar cuda error:

`[Exception|implicit_gemm_pair]indices=torch.Size([11829, 4]),bs=1,ss=[  41 1600 1408],algo=ConvAlgo.MaskImplicitGemm,ksize=[3, 3, 3],stride=[1, 1, 1],padding=[1, 1, 1],dilation=[1, 1, 1],subm=True,transpose=False
SPCONV_DEBUG_SAVE_PATH not found, you can specify SPCONV_DEBUG_SAVE_PATH as debug data save path to save debug data which can be attached in a issue.
Traceback (most recent call last):
  File "demo.py", line 118, in <module>
    main()
  File "demo.py", line 106, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/wen/Documents/PointCloudDet/OpenPCDet/pcdet/models/detectors/voxel_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/wen/Documents/Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/PointCloudDet/OpenPCDet/pcdet/models/backbones_3d/spconv_backbone.py", line 148, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/wen/Documents/Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/modules.py", line 137, in forward
    input = module(input)
  File "/home/wen/Documents/Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/conv.py", line 410, in forward
    raise e 
  File "/home/wen/Documents/Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/conv.py", line 401, in forward
    timer=input._timer)
  File "/home/wen/Documents/Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/ops.py", line 376, in get_indice_pairs_implicit_gemm
    mask_argsort_tv[j], stream)
RuntimeError: radix_sort: failed on 1st step: cudaErrorInvalidDeviceFunction: invalid device function
`

I tried cuda 9.2, 10.2, and 11.5 all getting the same error.
I am using a GM107GL [Quadro K620] GPU. I have read elsewhere that this error may be solved with a more powerful GPU, is it the case here?

Looking forward to your reply

MartinHahner, 2021-11-27:
@FindDefinition: What is the status on Maxwell GPU support?

FindDefinition, 2021-11-29:
@MartinHahner sm_50 now available in spconv-cu102>=2.1.16.

twn29004, 2021-12-02:
> Hi,
>
> When I run demo.py I got a similar cuda error:
>
> `[Exception|implicit_gemm_pair]indices=torch.Size([11829, 4]),bs=1,ss=[ 41 1600 1408],algo=ConvAlgo.MaskImplicitGemm,ksize=[3, 3, 3],stride=[1, [...]

Hi, have you solved the problem?

QAQAQAQAQAQ, 2021-12-02:
Sort of. Yes. 
The error message changed to:
`[Exception|implicit_gemm]feat=torch.Size([11829, 4]),w=torch.Size([16, 3, 3, 3, 4]),pair=torch.Size([27, 11829]),act=11829,issubm=True,istrain=False
SPCONV_DEBUG_SAVE_PATH not found, you can specify SPCONV_DEBUG_SAVE_PATH as debug data save path to save debug data which can be attached in a issue.
Traceback (most recent call last):
  File "demo.py", line 119, in <module>
    main()
  File "demo.py", line 107, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/wen/Documents/PointCloudDet/OpenPCDet/pcdet/models/detectors/voxel_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/PointCloudDet/OpenPCDet/pcdet/models/backbones_3d/spconv_backbone.py", line 148, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/modules.py", line 137, in forward
    input = module(input)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/conv.py", line 448, in forward
    input._timer)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/cuda/amp/autocast_mode.py", line 94, in decorate_fwd
    return fwd(*args, **kwargs)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/functional.py", line 197, in forward
    raise e 
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/functional.py", line 188, in forward
    is_train, is_subm, timer)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/ops.py", line 1116, in implicit_gemm
    stream=stream)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/algo.py", line 645, in tune_and_cache
    raise ValueError("can't find suitable algorithm for", op_type)
ValueError: ("can't find suitable algorithm for", <ConvOpType.kForward: 0>)

`

I used spconv-cu102==2.1.18
cuda==10.2
torch==1.10.0

I assumed I was using a different combination of packages so I also tried torch==1.4.0 & 1.6.0 (I did run setup.py again after that) but there seems to be a lot of dependency issues.

I see the [INSTALL.md](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md) was updated.
> You could also install latest spconv v2.x with pip, see the official documents of spconv.
Is there a prerequisite for PyTorch version or can I just use anything?

FindDefinition, 2021-12-03:
@QAQAQAQAQAQ fixed in cumm >= 0.2.6, you need to upgrade cumm by ```pip install -U cumm-cu102 -i https://pypi.python.org/simple``` after [build](https://github.com/FindDefinition/cumm/actions/runs/1533388129) finished.

QAQAQAQAQAQ, 2021-12-04:
Thank you for your fast feedback.
I updated cumm but now I am getting cuda error 710 now

`
this arch isn't supported!
/io/build/temp.linux-x86_64-3.6/spconv/build/src/cumm/conv/main/Simt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK/ConvKernel/ConvKernel_conv_kernel.cu:104: void cumm::conv::main::Simt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK::conv_kernel(cumm::conv::main::cpSimt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK::ConvParams): block: [6,0,0], thread: [0,0,0] Assertion `0` failed.

......
/io/build/temp.linux-x86_64-3.6/spconv/build/src/cumm/conv/main/Simt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK/ConvKernel/ConvKernel_conv_kernel.cu:104: void cumm::conv::main::Simt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK::conv_kernel(cumm::conv::main::cpSimt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK::ConvParams): block: [6,0,0], thread: [158,0,0] Assertion `0` failed.
/io/build/temp.linux-x86_64-3.6/spconv/build/src/cumm/conv/main/Simt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK/ConvKernel/ConvKernel_conv_kernel.cu:104: void cumm::conv::main::Simt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK::conv_kernel(cumm::conv::main::cpSimt_f32f32f32f32f32tnt_m32n128k16m32n32k8_200_C301LLLA1_SK::ConvParams): block: [6,0,0], thread: [159,0,0] Assertion `0` failed.
[Exception|implicit_gemm]feat=torch.Size([40000, 4]),w=torch.Size([16, 3, 3, 3, 4]),pair=torch.Size([27, 40000]),act=40000,issubm=True,istrain=False
SPCONV_DEBUG_SAVE_PATH not found, you can specify SPCONV_DEBUG_SAVE_PATH as debug data save path to save debug data which can be attached in a issue.
Traceback (most recent call last):
  File "demo.py", line 114, in <module>
    main()
  File "demo.py", line 102, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/wen/Documents/PointCloudDet/OpenPCDet/pcdet/models/detectors/voxel_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/PointCloudDet/OpenPCDet/pcdet/models/backbones_3d/spconv_backbone.py", line 148, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/modules.py", line 137, in forward
    input = module(input)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/conv.py", line 448, in forward
    input._timer)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/torch/cuda/amp/autocast_mode.py", line 94, in decorate_fwd
    return fwd(*args, **kwargs)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/functional.py", line 197, in forward
    raise e 
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/functional.py", line 188, in forward
    is_train, is_subm, timer)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/pytorch/ops.py", line 1116, in implicit_gemm
    stream=stream)
  File "/home/wen/Documents/EECS 498 Introduction to Algorithmic Robotics/HW3files/OpenPCDet/lib/python3.6/site-packages/spconv/algo.py", line 638, in tune_and_cache
    GemmMainUnitTest.stream_synchronize(stream)
RuntimeError: /io/build/temp.linux-x86_64-3.6/spconv/build/src/cumm/gemm/main/GemmMainUnitTest/GemmMainUnitTest_stream_synchronize.cc(11)
CUDA error 710
`
Thank you for your patience.

FindDefinition, 2021-12-04:
@QAQAQAQAQAQ this problem should be fixed in spconv 2.1.19. upgrade and try again.

QAQAQAQAQAQ, 2021-12-04:
Thank you so much!!! demo.py is working correctly right now.

Yaccoub, 2021-12-08:
I am facing the same error while using it on google colab 
[image]

with the following configuration: 
CUDA = V10.2.89
torch = 1.10.0+cu102
spconv = 2.1.20 
Thank you for your help !

FindDefinition, 2021-12-08:
@Yaccoub you are using K80 (sm_37) in colab which isn't supported by spconv prebuilts yet, will be added in spconv-cu102 2.1.21.

YomnaGamal, 2021-12-24:
> @Yaccoub you are using K80 (sm_37) in colab which isn't supported by spconv prebuilts yet, will be added in spconv-cu102 2.1.21.

I'm using spconv-cu111  2.1.21 with K80, torch 1.10.0+cu111, torchvision 0.11.1+cu111 and the same error appears to me too.  
[image]
any solution? 
Thanks.