open-mmlab/OpenPCDet #279: CLASS_AGNOSTIC parameter in ROI_HEAD

debortoli, 2020-09-04:
Hi, 

Thanks for releasing this great library. I am trying to use the PartA2 model with my custom data and notice that the ROI_HEAD has the CLASS_AGNOSTIC parameter set to True, meaning that when that head gets built, the class size is 1 (https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/models/detectors/detector3d_template.py#L159). 

I think I'm mis-understanding something because this seems to be the output for the network and I assume a dataset can have more than one class. Thanks!

jihanyang, 2020-09-08:
CLASS_AGNOSTIC NMS means doing nms for each class, so if you set this to false, nms function doesn't distinguish each class. In this regard, the number of classes can be regard as 1.

debortoli, 2020-09-09:
@jihanyang ok that makes sense thank you!

More what I was wondering about was how to get predicted bounding boxes with a softmax output, meaning the classification output would have a dimension # of classes. This information seems to be at the point-predicted classification stage but gets destroyed during the ROI_HEAD and I was wondering if that information is recoverable.

sshaoshuai, 2020-10-19:
You could record the classification scores in RPN stage and always track it until the final predictions. The codes currently do not have this feature.

closed 2020-10-20 by sshaoshuai