open-mmlab/OpenPCDet #280: Training PartA2 network

hcyoo93, 2020-09-06:
Hello @sshaoshuai,

Thank you for your great work!

I have trouble training PartA2 network.

Evaluating PartA2 and training PV-RCNN, PointRCNN, and SECOND work fine.

`x = self.conv_input(input_sp_tensor)` of UNetV2 gives **nan** values **for some batches**, and the nan value yields  0 max_overlap, 0 foreground and 0 background.

How can I resolve this issue? Did you have any similar experience? 

Here is the log. Please note that my data set is Nusecenes exported to KITTI.

`2020-09-07 00:43:50,426   INFO  **********************Start logging**********************
2020-09-07 00:43:50,426   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-09-07 00:43:50,426   INFO  cfg_file         /home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/tools/cfgs/kitti_models/PartA2.yaml
2020-09-07 00:43:50,426   INFO  batch_size       1
2020-09-07 00:43:50,426   INFO  epochs           85
2020-09-07 00:43:50,427   INFO  workers          8
2020-09-07 00:43:50,427   INFO  extra_tag        default
2020-09-07 00:43:50,427   INFO  ckpt             /home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/checkpoints/kitti/PartA2_7940.pth
2020-09-07 00:43:50,427   INFO  pretrained_model None
2020-09-07 00:43:50,427   INFO  launcher         none
2020-09-07 00:43:50,427   INFO  tcp_port         18888
2020-09-07 00:43:50,427   INFO  sync_bn          False
2020-09-07 00:43:50,427   INFO  fix_random_seed  False
2020-09-07 00:43:50,427   INFO  ckpt_save_interval 1
2020-09-07 00:43:50,427   INFO  local_rank       0
2020-09-07 00:43:50,427   INFO  max_ckpt_save_num 30
2020-09-07 00:43:50,427   INFO  merge_all_iters_to_one_epoch False
2020-09-07 00:43:50,427   INFO  set_cfgs         None
2020-09-07 00:43:50,427   INFO  max_waiting_mins 0
2020-09-07 00:43:50,427   INFO  start_epoch      0
2020-09-07 00:43:50,427   INFO  save_to_file     False
2020-09-07 00:43:50,427   INFO  cfg.ROOT_DIR: /home/hcyoo/anaconda3/envs/pcdet/OpenPCDet
2020-09-07 00:43:50,427   INFO  cfg.LOCAL_RANK: 0
2020-09-07 00:43:50,427   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-09-07 00:43:50,427   INFO  
cfg.DATA_CONFIG = edict()
2020-09-07 00:43:50,427   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-09-07 00:43:50,427   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2020-09-07 00:43:50,427   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2020-09-07 00:43:50,427   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-09-07 00:43:50,427   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-09-07 00:43:50,427   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-09-07 00:43:50,427   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2020-09-07 00:43:50,427   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2020-09-07 00:43:50,427   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2020-09-07 00:43:50,428   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-09-07 00:43:50,428   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-09-07 00:43:50,428   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-09-07 00:43:50,428   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:20', 'Pedestrian:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-09-07 00:43:50,428   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-09-07 00:43:50,428   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-09-07 00:43:50,428   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2020-09-07 00:43:50,428   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2020-09-07 00:43:50,428   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2020-09-07 00:43:50,428   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2020-09-07 00:43:50,428   INFO  
cfg.MODEL = edict()
2020-09-07 00:43:50,428   INFO  cfg.MODEL.NAME: PartA2Net
2020-09-07 00:43:50,428   INFO  
cfg.MODEL.VFE = edict()
2020-09-07 00:43:50,428   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2020-09-07 00:43:50,428   INFO  
[... 493 lines cut ...]
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/tools/train_utils/train_utils.py", line 93, in train_model
    dataloader_iter=dataloader_iter
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/tools/train_utils/train_utils.py", line 38, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/pcdet/models/__init__.py", line 30, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/hcyoo/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/pcdet/models/detectors/PartA2_net.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/hcyoo/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/pcdet/models/roi_heads/partA2_head.py", line 175, in forward
    targets_dict = self.assign_targets(batch_dict)
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/pcdet/models/roi_heads/roi_head_template.py", line 104, in assign_targets
    targets_dict = self.proposal_target_layer.forward(batch_dict)
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 33, in forward
    batch_dict=batch_dict
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 107, in sample_rois_for_rcnn
    sampled_inds = self.subsample_rois(max_overlaps=max_overlaps)
  File "/home/hcyoo/anaconda3/envs/pcdet/OpenPCDet/pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 159, in subsample_rois
    raise NotImplementedError
NotImplementedError
`

hcyoo93, 2020-09-07:
Changing version from "pytorch 1.5, spconv 1.2" to "pytorch1.1, spconv1.0" resolves the issue.
I think, anyone who has trouble with spconv can consider downgrading it

closed 2020-09-07 by hcyoo93

yustaub, 2020-12-09:
Hi, sir. Do you convert the nuscenes dataset to kitti format and trained with PartA2? May I ask how I can transform nuscenes to kitti, I want to train pv-rcnn on nuscenes but has no idea about it. Very appreciate for your reply. @hcyoo93

hcyoo93, 2020-12-09:
Hi, @yustaub 

Yes, I convert Nuscenes to KITTI since I want to do cross-evaluation of models on the dataset.

I think you can refer `https://github.com/nutonomy/nuscenes-devkit/blob/02e9200218977193a1058dd7234f935834378319/python-sdk/nuscenes/scripts/export_kitti.py`

yustaub, 2020-12-09:
Sorry to bother you again. Do you also change the categories of the original PartA2 config, or just use the original config? @hcyoo93

hcyoo93, 2020-12-12:
Hi, @yustaub

I think it depends on you.

You need to change anchor size, point cloud range, etc.
Note that the anchor size is the average instance size of the dataset.
Also, Nuscenes predict 360 degrees while KITTI only cares about front cam FOV.

you can refer to this paper for cross-dataset training and evaluation
https://arxiv.org/pdf/2005.08139.pdf

yustaub, 2020-12-13:
thanks for your reply! Appreciate!

JianyyuWang, 2022-08-25:
> Changing version from "pytorch 1.5, spconv 1.2" to "pytorch1.1, spconv1.0" resolves the issue. I think, anyone who has trouble with spconv can consider downgrading it

hi, i used your way to solve the problem,but failed. do you have some other methods?

lcc815, 2023-02-07:
> > Changing version from "pytorch 1.5, spconv 1.2" to "pytorch1.1, spconv1.0" resolves the issue. I think, anyone who has trouble with spconv can consider downgrading it
> 
> hi, i used your way to solve the problem,but failed. do you have some other methods?

same question! Ask for help.

Tottowich, 2023-08-21:
Did you find any other solution to 
```
OpenPCDet/tools/../pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 159, in subsample_rois
    raise NotImplementedError
```
@JianyyuWang @lcc815 ?

Weiquan1, 2024-09-19:
Any updates on this? I am currently stuck with the same error. I have tried different versions of spconv but still no different.