open-mmlab/OpenPCDet #281: Using evaluate.py - is it functional?

jacoblambert, 2020-09-07:
Since I see no mention of it in anywhere, I'm wondering if evaluate.py is functional?

First I had to modify it to use it at all, then when I tried using it at all on my custom data, I got bad results despite knowing the model is performing quite well.

So to test it, I took 20 ground truth labels, copied them and added fake scores between 0.9 and 1.0. I may be misunderstanding mAP (low number of examples makes the metric weird, but I have 'perfect' detection here), but it seems I should be getting reasonably high scores? For cars, I get 

`Car AP@0.70, 0.70, 0.70:
bbox AP:27.2727, 54.5455, 63.6364
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:27.27, 54.55, 63.64
`

Not even consistent between evaluations. Why 0 BEV and 3D, is it not running on those metrics? I don't see anything to do with calibration in the eval code, so shouldn't be an issue with reference frames. I'm lost. Is something wrong with this code?

jihanyang, 2020-09-08:
How did you fill the 2D information with your data when evaluation?

jacoblambert, 2020-09-09:
@jihanyang 
On custom data I set the 2D bounding box to "0 0 50 50" and set all other fields to 0.
But as I explained above, even using the ground truth as "prediction labels", the score is low.

josy43, 2020-09-14:
I have the same problem, don´t know why

thaotrongtran, 2020-09-15:
I am also trying to figure out how to tweak the evaluation code to be useful with custom datasets. Still getting 0 on 3D metric.

josy43, 2020-09-15:
I still didnt figured it out. I wouldnt recommend to use the evaluation because there is a massive bug.

hoangduyloc, 2020-09-16:
@jacoblambert  
Hello, Have you ever evaluated on the official KITTI benchmark, Can I set 2D bb -KITTI required to (0, 0, 50, 50) since my detection is only BEV? Does it affect the evaluation result? Thank you

jacoblambert, 2020-09-17:
@hoangduyloc if you are using test.py then I think there is no problem.

This thread is about evaluate.py, it seems bugged. I've tried to disentangle the code but could not locate the problem yet.

hoangduyloc, 2020-09-17:
> @hoangduyloc if you are using test.py then I think there is no problem.
> 
> This thread is about evaluate.py, it seems bugged. I've tried to disentangle the code but could not locate the problem yet.

Oh, yes. I do if using for the test so we can set arbitrary. But when evaluating on Official KITTI Benchmark, It requires accurate 2D BB information. But the reason is I can't have accurate Height value because I predict on BEV image so it affects my evaluation.

jacoblambert, 2020-09-18:
Using the official (offline) KITTI cpp evaluation code, I confirmed there was nothing wrong with my labels. For example using the "fake" labels created from the ground truth ones (irrelevant classes removes, added scores) everything seeems fine:
```
start to eval image 2D bounding boxes... 
car_detection AP: 100.000000 100.000000 100.000000
  start to compute aos ... 
car_orientation AP: 100.000000 100.000000 100.000000
start to eval bird's eye view bounding boxes... 
car_detection_ground AP: 100.000000 100.000000 100.000000
start to eval 3D bounding boxes... 
car_detection_3d AP: 100.000000 100.000000 100.000000
```

and then evaluate.py results:
```
Car AP@0.70, 0.70, 0.70:
bbox AP:0.0425, 0.0535, 0.0719
bev  AP:0.0085, 0.0126, 0.0155
3d   AP:0.0043, 0.0052, 0.0066
aos  AP:0.03, 0.04, 0.05
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:0.0117, 0.0147, 0.0198
bev  AP:0.0023, 0.0035, 0.0043
3d   AP:0.0012, 0.0014, 0.0018
aos  AP:0.01, 0.01, 0.01
Car AP@0.70, 0.50, 0.50:
bbox AP:0.0425, 0.0535, 0.0719
bev  AP:0.0273, 0.0366, 0.0466
3d   AP:0.0150, 0.0222, 0.0284
aos  AP:0.03, 0.04, 0.05
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:0.0117, 0.0147, 0.0198
bev  AP:0.0075, 0.0101, 0.0128
3d   AP:0.0041, 0.0061, 0.0078
aos  AP:0.01, 0.01, 0.01
```

I can't figure out what's wrong with the evaluation.py code that is different from the normal evaluation...

jacoblambert, 2020-09-22:
Since I couldn't solve this in PCDet repo, I'm resorting to using official eval code. I made some script to transfer whatever your format is to KITTI format, then you can run the cpp code simply: https://github.com/jacoblambert/label_to_kitti_format

jhultman, 2020-09-28:
@jacoblambert I think the problem you mention is the same as the one discussed [here](https://github.com/poodarchu/Det3D/issues/75) and [here](https://github.com/traveller59/kitti-object-eval-python/issues/4). The rotated IOU algorithm in the eval library cannot handle the case where the orientations are identical (this is a bug of course).

closed 2022-03-08 by github-actions[bot]