open-mmlab/OpenPCDet #284: error: ‘AT_CHECK’ was not declared in this scope

McDifference, 2020-09-11:
OS: Ubuntu 16.04
Python: 3.6
CUDA: 10.0
PyTorch: 1.2

After I run the command "python setup.py develop", the following errors occured. I have also switched CUDA to 10.1 and tried PyTorch versions of 1.3, 1.4, 1.5, 1.6, but the errors remained the same.

```
In file included from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/Tensor.h:3:0,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/Context.h:4,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/ATen.h:5,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from /home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:7:
/home/user/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:354:7: note: declared here
   T * data() const {
       ^
/home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp: In function ‘int nms_gpu(at::Tensor, at::Tensor, float)’:
/home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:14:39: warning: ‘at::DeprecatedTypeProperties& at::Tensor::type() const’ is deprecated: Tensor.type() is deprecated. Instead use Tensor.options(), which in many cases (e.g. in a constructor) is a drop-in replacement. If you were using data from type(), that is now available from Tensor itself, so instead of tensor.type().scalar_type(), use tensor.scalar_type() instead and instead of tensor.type().backend() use tensor.device(). [-Wdeprecated-declarations]
 #define CHECK_CUDA(x) AT_CHECK(x.type().is_cuda(), #x, " must be a CUDAtensor ")
                                       ^
/home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:16:24: note: in expansion of macro ‘CHECK_CUDA’
 #define CHECK_INPUT(x) CHECK_CUDA(x);CHECK_CONTIGUOUS(x)
                        ^
/home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:84:5: note: in expansion of macro ‘CHECK_INPUT’
     CHECK_INPUT(boxes);
     ^
In file included from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/Tensor.h:3:0,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/Context.h:4,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/ATen.h:5,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from /home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:7:
/home/user/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:268:30: note: declared here
   DeprecatedTypeProperties & type() const {
                              ^
/home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:14:80: error: ‘AT_CHECK’ was not declared in this scope
 #define CHECK_CUDA(x) AT_CHECK(x.type().is_cuda(), #x, " must be a CUDAtensor ")
                                                                                ^
/home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:16:24: note: in expansion of macro ‘CHECK_CUDA’
 #define CHECK_INPUT(x) CHECK_CUDA(x);CHECK_CONTIGUOUS(x)
                        ^
/home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:84:5: note: in expansion of macro ‘CHECK_INPUT’
     CHECK_INPUT(boxes);
     ^
/home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:88:50: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
     const float * boxes_data = boxes.data<float>();
                                                  ^
In file included from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/Tensor.h:3:0,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/Context.h:4,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/ATen/ATen.h:5,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /home/user/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from /home/user/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:7:
/home/user/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:354:7: note: declared here
   T * data() const {
       ^
/home/k/workspace/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms.cpp:89:40: warning: ‘T* at::Tensor::data() const [with T = long int]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
     long * keep_data = keep.data<long>();
```

McDifference, 2020-09-11:
Problem solved.
The cause of this problem is that OpenPCDet currently does not support PyTorch1.6.
Actually my first build was with CUDA10.1 and PyTorch1.6, and some meta files were generated in the build folder as well as the pcdet.egg-info. I forgot to delete these files before I switched to PyTorch1.2-1.5.

Now I have deleted the whole repo and cloned it again. Build is successful with CUDA10.1 and PyTorch1.5.

closed 2020-09-11 by McDifference

pfollmann, 2021-07-05:
I think another option would be to replace AT_CHECK with TORCH_CHECK for later Pytorch versions, see [here](https://github.com/pytorch/pytorch/pull/20321)