open-mmlab/OpenPCDet #285: Evaluation bug?

alexa19lab, 2020-09-11:
so when you eval overlap of gt and pred 3d bboxes in [this file](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/kitti/kitti_object_eval_python/eval.py), usinng function `<d3_box_overlap_kernel>` , 
how are the input boxes encodes? 

According to the code, they are encodes with x, z, y , dx, dz, dy, yaw.  But shouldn't it be x,y,z,dx,dy,dz,yaw?

thaotrongtran, 2020-09-16:
How do you figure out this part? [x, z, y , dx, dz, dy, yaw]

alexa19lab, 2020-09-17:
> How do you figure out this part? [x, z, y , dx, dz, dy, yaw]

Because in the function 'd3_box_overlap_kernel', they use 
python
'''
iw = (min(boxes[i, 1], qboxes[j, 1]) - max(
                    boxes[i, 1] - boxes[i, 4], qboxes[j, 1] - qboxes[j, 4]))
'''
So I assume boxes[i, 1] and boxes[i, 4] corresponds to 'z' and 'dz'  respectively?  Then I guess boxes[i, 0] and boxes[i, 2] should correspond to 'x' and 'y'? So its '[x, z, y , dx, dz, dy, yaw]'?

c7huang, 2020-11-05:
I agree there may be a bug in the current KITTI eval code.

The evaluation code for KITTI seems to be adopted from [traveller59/kitti-object-eval-python](https://github.com/traveller59/kitti-object-eval-python) which is used in SECOND.pytorch as well. By default it assumes the `z_axis` is encoded in the second position of the regression output `[x, z, y, ...]`, but in PCDet it's `[x, y, z, ...]`. 

In the newer version of the eval code they support user-defined `z_axis` and `z_center` parameter. Instead of calculating `iw` using
```
iw = (min(boxes[i, 1], qboxes[j, 1]) - max(boxes[i, 1] - boxes[i, 4], qboxes[j, 1] - qboxes[j, 4]))
```
It is changed to
```
min_z = min(
      boxes[i, z_axis] + boxes[i, z_axis + 3] * (1 - z_center),
      qboxes[j, z_axis] + qboxes[j, z_axis + 3] * (1 - z_center))
max_z = max(
      boxes[i, z_axis] - boxes[i, z_axis + 3] * z_center,
      qboxes[j, z_axis] - qboxes[j, z_axis + 3] * z_center)
iw = min_z - max_z
```

**A fix may be to update `kitti_object_eval_python` and specify the correct `z_axis` and `z_center` values.**

See [change log](https://github.com/traveller59/kitti-object-eval-python/commit/0c4c45d2932cdf80a441d8d935158828d403da93#diff-0d966d6c14a6b78bc80ed395e2fe33e014e91bcaafdde33608359324e66a914b) and [SECOND.pytorch's KITTI eval code](https://github.com/traveller59/second.pytorch/blob/3aba19c9688274f75ebb5e576f65cfe54773c021/second/data/kitti_dataset.py#L109)

c7huang, 2020-11-05:
In the output prediction dictionary for KITTI dataset, there are a few fields that are used to encode the bounding boxes, namely
- `pred_boxes`
- `location`
- `dimensions`
- `rotation_y`

`pred_boxes` seems to be encoded in `[x,y,z,l,w,h,r]`, but the evaluation code uses `location`, `dimensions` and `rotation_y` to construct the boxes, **not**  `pred_boxes`:
```
# From `eval.py` calculate_iou_partly
loc = np.concatenate([a["location"] for a in dt_annos_part], 0)
dims = np.concatenate([a["dimensions"] for a in dt_annos_part], 0)
rots = np.concatenate([a["rotation_y"] for a in dt_annos_part], 0)
dt_boxes = np.concatenate([loc, dims, rots[..., np.newaxis]], axis=1)
```

These fields are converted to KITTI format in `generate_prediction_dicts`, and thus the axis should match  `[x,z,y,l,h,w,r]`.
```
# From `kitti_dataset.py` generate_prediction_dicts
pred_boxes_camera = box_utils.boxes3d_lidar_to_kitti_camera(pred_boxes, calib)
...
pred_dict['dimensions'] = pred_boxes_camera[:, 3:6]
pred_dict['location'] = pred_boxes_camera[:, 0:3]
pred_dict['rotation_y'] = pred_boxes_camera[:, 6]
```

So the KITTI evaluation results should be valid.

alexa19lab, 2020-11-06:
> In the output prediction dictionary for KITTI dataset, there are a few fields that are used to encode the bounding boxes, namely
>
> * `pred_boxes`
> * `location`
> * `dimensions`
> * `rotation_y` [...]

Thanks! Solves my problem!

closed 2020-11-06 by alexa19lab