open-mmlab/OpenPCDet #288: The nuscenes models do not match the config file

jaylee7, 2020-09-16:
python demo.py --cfg_file cfgs/nuscenes_models/cbgs_pp_multihead.yaml --ckpt pp_multihead_nds5823_updated.pth
&&
python demo.py --cfg_file cfgs/nuscenes_models/cbgs_second_multihead.yaml --ckpt cbgs_second_multihead_nds6229_updated.pth

both will report error！

MartinHahner88, 2020-09-16:
There are already other threads about this, https://github.com/open-mmlab/OpenPCDet/issues/257 and https://github.com/open-mmlab/OpenPCDet/issues/195.

jaylee7, 2020-09-16:
Thank you so much！

jaylee7, 2020-09-16:
I found the real reason.

In the **kitti_dataset.yaml**:
'''
POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}
'''

But in the **nuscenes_dataset.yaml**:
'''
POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
    src_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
}
'''

So if the input point cloud is N x 4 dimensions, it will report error. We need to change the input point cloud to N x 5!

jaylee7, 2020-09-16:
I'm not familiar with nuscenes, so I don't know what the correct timestamp is. I just set it to 0. It can works properly. So who can tell me the correct value?

And another question, the **DEMO.md** say: " If you have the intensity information, you should normalize them to [0, 1]."
But I found if the intensity is [0, 255], it has better performance when using pp_multihead_nds5823_updated.pth. Why?

MartinHahner88, 2020-09-17:
> I'm not familiar with nuscenes, so I don't know what the correct timestamp is. I just set it to 0. It can works properly. So who can tell me the correct value?

The nuScenes point clouds are way less dense because they used a cheaper LiDAR sensor to collect the dataset compared to the one they used to collect the KITTI dataset.
In order to make the nuScenes point cloud denser, a common technique is to use multiple sweeps (scans) at once, so combining usually 10 of them into a single denser point cloud. During this process, the time delta of the individual sweeps to the main (current) sweep is saved into the 5th feature, so there is no "correct value" in the DEMO scenario here and that's why setting it to 0 works. In that way, you are basically telling the network that all points belong to the current sweep.

jaylee7, 2020-09-18:
Thank you very much for your detailed reply, which perfectly solved my doubts.

closed 2020-09-18 by jaylee7

jaylee7, 2020-09-23:
As for the second question
I found some information about the nuScenes dataset:
1）The z-axis origin of your point cloud coordinate should be about 1.84m above the ground surface
2）The lidar intensity range is [0,100]
I hope it is useful for others.

MartinHahner88, 2020-09-23:
@jaylee7 
> The lidar intensity range is [0,100]

Where did you get this information?

jaylee7, 2020-09-24:
@MartinHahner88 
Sorry, I didn't check this information. The correct range is[0,255].
Thank you for your correction.