open-mmlab/OpenPCDet #289: About the given results

gardenbaby, 2020-09-17:
Hi, I wonder the given results on KITTI val set still use 11 recall positions or  40 recall positions. Thanks.

hoangduyloc, 2020-09-17:
They updated to use an average of 40 samples from [0,1] since  08.10.2019. You can see at Note here: http://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d#:~:text=The%203D%20object%20detection%20benchmark,methods%20we%20compute%20average%20precision.

gardenbaby, 2020-09-17:
> They updated to use an average of 40 samples from [0,1] since 08.10.2019. You can see at Note here: http://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d#:~:text=The%203D%20object%20detection%20benchmark,methods%20we%20compute%20average%20precision.

But it seems that the given results of baselines in file [README.md](https://github.com/open-mmlab/OpenPCDet/blob/master/README.md) still use 11 recall positions.

MartinHahner88, 2020-09-17:
To the best of my knowledge, [README.md](https://github.com/open-mmlab/OpenPCDet/blob/master/README.md) unfortunately still lists **mAP_11** with 11 recall positions.

gardenbaby, 2020-09-17:
@MartinHahner88 
Thank you.

closed 2020-10-19 by sshaoshuai