open-mmlab/OpenPCDet #290: How to project 3D BBoxes from prediction to 2D BBoxes ?

hoangduyloc, 2020-09-17:
Anyone can show me how can I project 3D BB from prediction to 2D BB accurately way?

Bynow76, 2020-09-18:
I have the same question

hoangduyloc, 2020-09-18:
@Bynow76 
Hello, I figured out the solution. When you do the projection: 3D coordinate project into the image plane [8, 3] --> [8, 2], after that find the min, max value along each column so that this rectangle will cover the object.

import torch

Coord_3D = torch.tensor(
[[672.59971903, 172.84996374],
 [654.37659384, 172.84996087],
 [6585.47537053, 172.84970659],
 [677.84173596, 172.84970984],
 [672.59971903, 190.10340984],
 [654.37659384, 1.11569124],
 [658.47537053, 191.20237318],
 [677.84173596, 191.18849741]])

Coord_2D = torch.cat([torch.min(A, dim=1)[0], torch.max(A, dim=1)[0]], dim=1)
print(Coord_2D)

Bynow76, 2020-09-19:
thanks a lot!

closed 2020-10-19 by sshaoshuai