open-mmlab/OpenPCDet #291: What's the meaning of the shown results?

3bobo, 2020-09-17:
Thanks for this work. I noticed that there have been shown some results of networks. However, when I test the pretrained model, Part-A2_free for example, after `python test.py --cfg_file cfgs/kitti_models/PartA2_free.yaml --ckpt ../check/PartA2_free_7872.pth`, I only get these results:
```
INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:90.7064, 89.1032, 88.7695
bev  AP:90.0969, 86.7886, 84.5979
3d   AP:89.1192, 78.7253, 77.9830
aos  AP:90.68, 89.00, 88.64
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:95.9395, 91.7175, 89.6303
bev  AP:92.8412, 88.1545, 86.1618
3d   AP:91.6842, 80.3118, 78.1013
aos  AP:95.90, 91.61, 89.50
Car AP@0.70, 0.50, 0.50:
bbox AP:90.7064, 89.1032, 88.7695
bev  AP:90.6940, 89.3172, 89.1733
3d   AP:90.6907, 89.2826, 89.1190
aos  AP:90.68, 89.00, 88.64
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:95.9395, 91.7175, 89.6303
bev  AP:95.9649, 94.0122, 93.6627
3d   AP:95.9425, 93.8626, 91.9144
aos  AP:95.90, 91.61, 89.50
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:80.2816, 75.6527, 71.7540
bev  AP:74.8279, 69.3140, 64.5455
3d   AP:70.3085, 65.9922, 60.4777
aos  AP:78.93, 74.03, 69.88
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:80.8329, 76.9968, 73.3469
bev  AP:75.4813, 69.8351, 64.4979
3d   AP:72.3584, 66.4026, 60.0709
aos  AP:79.32, 75.20, 71.20
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:80.2816, 75.6527, 71.7540
bev  AP:83.7051, 82.0240, 79.0633
3d   AP:83.7003, 81.9823, 78.9359
aos  AP:78.93, 74.03, 69.88
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:80.8329, 76.9968, 73.3469
bev  AP:87.3005, 84.9858, 80.3568
3d   AP:87.2962, 84.9335, 79.6124
aos  AP:79.32, 75.20, 71.20
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:89.4204, 81.0870, 77.1343
bev  AP:88.7514, 76.2628, 73.6792
3d   AP:87.6469, 74.2926, 69.9137
aos  AP:89.31, 80.57, 76.60
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:94.0643, 82.9229, 79.0304
bev  AP:93.2320, 78.4984, 73.9288
3d   AP:91.9237, 75.3263, 70.5746
aos  AP:93.90, 82.36, 78.47
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:89.4204, 81.0870, 77.1343
bev  AP:88.8460, 78.6822, 74.6989
3d   AP:88.8460, 78.6822, 74.6989
aos  AP:89.31, 80.57, 76.60
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:94.0643, 82.9229, 79.0304
bev  AP:93.3543, 80.3345, 76.5469
3d   AP:93.3543, 80.3345, 76.5469
aos  AP:93.90, 82.36, 78.47
```
I'm wondering which line result has the same meaning as  `78.72, 65.99, 74.29` of 

[image]

For example, is the 78.72 is the data of  `Car AP_R40@70/3d/3-rd col `? By the way, could I get the three different diffiulty results by runing `test.py` ? Thanks !

closed 2020-09-18 by 3bobo

451578928, 2020-09-22:
Hello~ I got the same confusion about the results, have you solved this problem?

3bobo, 2020-09-23:
@451578928 it seems that you should use `AP_R40` to compare, for example, `AP_R40@0.70 0.70 0.70` for `Car`, and the three columns are three different difficulty results respectively.

451578928, 2020-09-23:
I thought so at first, but I downloaded the official model. The result of running PV-RCNN on my device is:
```
Car AP@0.70, 0.70, 0.70:
bbox AP:96.2470, 89.4992, 89.2430
bev  AP:90.0894, 87.9004, 87.4072
3d   AP:89.3476, 83.6901, 78.7028
aos  AP:96.22, 89.39, 89.07
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:98.2662, 94.4208, 92.2765
bev  AP:93.0239, 90.3255, 88.5319
3d   AP:92.1047, 84.3605, 82.4830
aos  AP:98.25, 94.26, 92.07
Car AP@0.70, 0.50, 0.50:
bbox AP:96.2470, 89.4992, 89.2430
bev  AP:96.2811, 89.4982, 89.2886
3d   AP:96.2342, 89.4774, 89.2535
aos  AP:96.22, 89.39, 89.07
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:98.2662, 94.4208, 92.2765
bev  AP:98.2607, 94.5896, 94.4319
3d   AP:98.2422, 94.5277, 94.3272
aos  AP:98.25, 94.26, 92.07
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:73.1477, 68.0799, 64.3542
bev  AP:65.1821, 59.4169, 54.5101
3d   AP:63.1230, 54.8428, 51.7816
aos  AP:67.84, 62.49, 58.73
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:73.6837, 68.2715, 64.3622
bev  AP:65.9365, 58.5166, 54.1258
3d   AP:62.7110, 54.4902, 49.8798
aos  AP:67.82, 62.17, 58.07
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:73.1477, 68.0799, 64.3542
bev  AP:76.2555, 71.8445, 69.4931
3d   AP:76.2398, 71.8001, 69.4345
aos  AP:67.84, 62.49, 58.73
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:73.6837, 68.2715, 64.3622
bev  AP:78.2616, 73.1740, 69.9717
3d   AP:78.2458, 73.0349, 69.8725
aos  AP:67.82, 62.17, 58.07
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:96.1222, 81.3613, 76.4936
bev  AP:88.5292, 73.3251, 70.3690
3d   AP:86.0637, 69.4789, 64.5046
aos  AP:95.98, 81.07, 76.17
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:97.1514, 82.4114, 78.2196
bev  AP:93.4584, 74.5322, 70.1025
3d   AP:89.1011, 70.3809, 66.0168
aos  AP:97.04, 82.11, 77.88
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:96.1222, 81.3613, 76.4936
bev  AP:95.0655, 78.2760, 73.3191
3d   AP:95.0655, 78.2670, 73.3121
aos  AP:95.98, 81.07, 76.17
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:97.1514, 82.4114, 78.2196
bev  AP:96.2319, 79.1335, 75.8222
3d   AP:96.2319, 79.1278, 75.7990
aos  AP:97.04, 82.11, 77.88
```
It's very different from the official result：**

PV-RCNN | ~5 hours | 83.61 | 57.90 | 70.47 | model-50M
-- | -- | -- | -- | -- | --

**

3bobo, 2020-09-23:
@451578928 , it seems that your data is close to the official result, your results show `84.3605, 54.4902, 70.3809 `for `Car, Pedestrian and Cyclist` respectively, which has a little but not too much different from the official results. All official results are under moderate difficulty.

451578928, 2020-09-23:
I got it~ Thank you very much~ love you~