open-mmlab/OpenPCDet #293: The great different performance on the evaluation  by "train.py" or "test.py"

happinesslz, 2020-09-20:
@sshaoshuai  
Thanks for your great work. I find that the all results of "train with eval" by running "train.py" are much higher than running "test.py", which is very confused to me.
For example, I train with eval by runnning `CUDA_VISIBLE_DEVICES=0,1 python -m torch.distributed.launch --nproc_per_node=2 train.py --launcher pytorch --tcp_port 18887 --cfg_file ./cfgs/kitti_models/pointrcnn.yaml --epochs 80 --extra_tag  pointrcnn_two_gpu_epoch_80` .

On the 72th epoch, I obtain the following results by the way of "train with eval":

Car AP@0.70, 0.70, 0.70:
bbox AP:90.5989, 89.4011, 88.7285
bev  AP:89.7376, 87.5282, 86.7808
3d   AP:88.2964, 78.3302, 77.7414
aos  AP:90.59, 89.30, 88.54
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:96.4335, 92.6285, 90.3194
bev  AP:93.0822, 89.0516, 86.8855
3d   AP:**89.3509, 80.3378, 78.0754**
aos  AP:96.42, 92.51, 90.14
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:73.4262, 66.0558, 62.3488
bev  AP:69.1679, 61.4667, 54.9045
3d   AP:63.8616, 55.7287, 51.3790
aos  AP:70.70, 62.89, 58.79
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:73.7187, 66.5245, 61.0004
bev  AP:68.9220, 61.1919, 52.9619
3d   AP:**65.2837, 56.3750, 49.4000**
aos  AP:70.78, 62.85, 57.27
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:89.7579, 77.7268, 75.7171
bev  AP:87.0327, 73.3521, 67.6928
3d   AP:86.3409, 72.2132, 66.9524
aos  AP:89.68, 77.27, 75.24
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:95.2092, 79.8456, 75.3625
bev  AP:91.9242, 74.4378, 69.8577
3d   AP:**90.9539, 72.0658, 67.5630**
aos  AP:95.11, 79.35, 74.86

However, I obtain the  72th epoch results by running  `CUDA_VISIBLE_DEVICES=0,1  python -m torch.distributed.launch --nproc_per_node=2 test.py  --launcher pytorch --cfg_file ./cfgs/kitti_models/pointrcnn.yaml --eval_all  --start_epoch 70   --extra_tag  pointrcnn_two_gpu_epoch_80`
The 72th epoch reulsts are following:

Car AP@0.70, 0.70, 0.70: 
bbox AP:90.7030, 89.4858, 88.7473
bev  AP:89.8970, 86.7371, 87.0516
3d   AP:88.1422, 77.6788, 76.7019
aos  AP:90.69, 89.38, 88.55
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:96.5968, 92.8132, 90.4110
bev  AP:93.2328, 87.1699, 86.8551
3d   AP:**89.4210, 79.6598, 77.4169**
aos  AP:96.57, 92.70, 90.20
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:73.7552, 65.2666, 57.6943
bev  AP:65.0447, 56.0764, 52.4499
3d   AP:63.0498, 54.1527, 46.6872
aos  AP:70.17, 61.60, 54.45
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:74.0334, 66.0200, 58.6183
bev  AP:66.9919, 57.1883, 51.0793
3d   AP:**63.5069, 53.8741, 46.5959**
aos  AP:70.22, 61.96, 54.76
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:89.2901, 70.7624, 69.9738
bev  AP:86.9400, 68.0235, 66.6615
3d   AP:85.1426, 66.2023, 59.2804
aos  AP:89.23, 70.63, 69.78
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:95.0044, 73.6662, 69.0920
bev  AP:90.0005, 68.9522, 66.2717
3d   AP:**86.3589, 65.2822, 61.0447**
aos  AP:94.89, 73.48, 68.93

How to slove it? thanks!

closed 2020-09-23 by happinesslz

xjjs, 2020-09-26:
@happinesslz 
> 
> How to slove it? thanks!

I have met the same issue.  Have you solved it?

xjjs, 2020-09-27:
Maybe the random seed is the reason

debortoli, 2020-10-03:
@happinesslz did you figure this out? I seem to be having similar problems. Thanks!