open-mmlab/OpenPCDet #295: About the box coder

gardenbaby, 2020-09-21:
Hi team,

I wonder why `PointResidualCoder` of code size 8(with class) is utilized for POINT_HEAD while `ResidualCoder` of code size 7 is utilized for ROI_HEAD.

Will the classes encoded in PointResidualCoder cause an adverse effect on the performance of the first stage by  increasing the complexity of the box coder ? 

Besides, is it necessary to learn box scores when there exsits the classification branch for point scores, since boxes are generated from points for POINT_HEAD?

Thanks.

sshaoshuai, 2020-10-19:
The code size 8 of `PointResidualCoder` is for (cx, cy, cz, dx, dy, dz, cos, sin), instead of `with class`. 

The point segmentation scores is not well matched with the box quality, so we need to do RoI Pooling for predicting better box score.

gardenbaby, 2020-10-20:
@sshaoshuai Thank you. Later I found that sin-cos angle coding is already used in `PointResidualCoder` with the extra parameter `gt_class`(in `encode()`) or `pred_class`(in `decode()`) for choosing the corresponding mean box size.

closed 2020-10-20 by sshaoshuai