open-mmlab/OpenPCDet #301: The difference between yours and the original version of SECOND

SiyuanHuang95, 2020-10-01:
Hi, first of all, thanks for your sharing of this wonderful work.

I noticed that you have re-implemented some other detectors, like SECOND. And in this [Issue](https://github.com/open-mmlab/OpenPCDet/issues/30#issuecomment-610158032), you have mentioned that there is no difference w.r.t the data augmentation. But what for other modules? I am a newbie in DL for PC detection, and would like to dive into one detector and try to reproduce it to get some deep insights.

BTW, I have found you did some comparisons in accuracy, but what about run-time, like FPS?

Best

sshaoshuai, 2020-10-19:
The implementation of SECOND (v1.5) should be similar with the original repo. For the inference time, you could test it with the provided model on your own machine.

SiyuanHuang95, 2020-10-20:
Thanks for your reply. And I will close this issue if nobody else has relevant question.

closed 2020-10-20 by SiyuanHuang95